aws/amazon-freertos #1: Support for ESP8266

JasonSwindle, 2017-11-29:
Howdy,

Support for ESP8266 would be wonderful, as it would open up this to a large group of IOT devices.

ourairquality, 2017-12-01:
The esp-open-rtos framework for the esp8266 uses a recent version of FreeRTOS and https://github.com/SuperHouse/esp-open-rtos/pull/500 updates this to use this code. But the IP network layer uses lwip, and newlib is used for the C library, and there are a few ssl options being explored, so it is not using all of the stack here.

adjervis, 2017-12-07:
Thank you for the valuable feedback. Such requests are important to us and we will take them into consideration in prioritizing our support for the ESP8266 device. In the meantime, you might find this link helpful: https://github.com/SuperHouse/esp-open-rtos.

gaborgsomogyi, 2018-06-01:
Any update on this?

aggarg, 2018-06-06:
Thank you for checking. The Amazon FreeRTOS does not support ESP8266 as of now. However we do support ESP32.

We will keep you updated if anything changes.

Thanks.

closed 2018-06-06 by aggarg

jasonbrowan, 2019-10-14:
Do you have plan to port to ESP8266?
Thanks.

cr8myf8, 2020-02-11:
+1 for support on the ESP8266

n-sviridenko, 2021-02-11:
+1, still relevant

rokkino, 2022-07-12:
hiii