aws/amazon-freertos #3: Support for Espressif ESP32

sethvoltz, 2017-11-30:
Similar to #1, support for the ESP32 micro-controller opens up a large number of inexpensive edge devices that come stock with wifi, bluetooth and crypto.

denyssene, 2017-12-01:
I think there's a port of freertos that works in esp32, but it's unofficial.

locomuco, 2017-12-01:
espressif IDF utilizes FreeRTOS already

see [here](https://github.com/espressif/esp-idfl)

sethvoltz, 2017-12-01:
I assume there is still additional work to be done for Amazon to validate their libs but that’s great that it already supports FreeRTOS in general.

adjervis, 2017-12-07:
Thank you for the valuable feedback. Such requests are important to us and we will take them into consideration in prioritizing our support for the ESP32 device.

gtech12, 2017-12-09:
Is there support for ESP32 or will it take time ?

adjervis, 2017-12-10:
We value your input, and will consider support for ESP32 in our future plans.

mrmonteith, 2018-03-09:
Any update on this?

prasadvya, 2018-03-09:
Thanks again for your interest. ExpressIf and Amazon FreeRTOS (AFR) are currently collaborating on a project to allow AFR to support ESP32 ([see here](https://www.espressif.com/en/media_overview/news/espressifs-esp32-support-amazon-freertos)). We will keep you appraised about when the port to ESP32 is available.

denyssene, 2018-03-09:
Nice. Ordering new esp32 boards now!

Alutun, 2018-03-28:
I hope it will come out soon !

erkki, 2018-04-03:
Any comments on the future direction of this? Will you be porting ESP32 specific (SMP related) FreeRTOS changes to AFR or pushing for ExpressIf to integrate AFR related bits into their SDK? This impacts future planning as their current AWS IoT integration is based on the aws-iot-device-sdk-embedded-C which AFR is not using (as far as I can see. Any rationale on this?)

ourairquality, 2018-04-03:
Would also be curious if FreeRTOS will integrate the SMP and other changes, but also if the ESP code will still use the lwip stack or if the goal is to use the AFR tcp/ip stack?

Little off topic, but for the esp8266 recent changes support an extra 16k of iram that makes SSL more practical, and might make this a possible limited target too.

'SSL maximum fragment size' support on the Amazon servers is badly needed and is there any chance of  changes on that side too to better support the IoT using limited devices?

awsota, 2018-04-07:
We are currently working with Espressif to launch support for ESP32. We are using a single core only, but the SMP version from Espressif may be compatible, if needed for your use case.

Alutun, 2018-04-17:
Same question here =) Would be nice to have an ETA even not precise but we can get an idea, I got an huge project coming, If i start with espressif SDK and three weeks later it's release I wlll waste time.

ravibhagavandas, 2018-05-16:
Support for Espressif ESP32 is now available with the release of Amazon FreeRTOS 1.2.5.

denyssene, 2018-05-16:
That's great news!

ravibhagavandas, 2018-05-16:
No, dual core is not supported at this time with Amazon FreeRTOS for ESP32.

ravibhagavandas, 2018-05-16:
Adding to the above comment, Espressif has developed **ESP-IDF FreeRTOS** which is a modified version of Vanilla FreeRTOS with support for SMP. 
It might be possible in your case to combine the **ESP**-IDF version of the FreeRTOS kernel with Amazon FreeRTOS libraries to run FreeRTOS on dual cores and securely connect to AWS IOT.

anujdeshpande, 2018-05-17:
@ravibhagavandas ESP-IDF also has support for AWS IoT using the [AWS IoT SDK for C](https://github.com/aws/aws-iot-device-sdk-embedded-C)

cagdasdoner, 2018-05-18:
Anybody else having trouble with compiling demo code -because of current lwip stack of IDF-? :hand: 
i.e. lwip/ip_addr.h:85:39: error: 'ip_addr_t {aka const struct ip4_addr}' has no member named 'u_addr'

jllorenzo, 2018-05-19:
cagdasdoner (If IDF_PATH is set then is recommend that you remove it before building Amazon FreeRTOS

pavanshah2001, 2018-06-02:
All the demos run properly, except the OTA update demo, it give following error ,undefined reference to `vStartOTAUpdateDemoTask'.
I just executed all demos, by commenting and commenting demos in aws_demo_runner.c

Looks like something minor, kindly help

Noggin01, 2018-06-02:
Amazon doesn't support lwIP which is required for OTA updates.  You'll need to update the OS to support lwIP or update OTA to use the sockets that Amazon is using.  I suspect updating the OTA would be easier.

I've switched to using the official ESP-IDF instead of a:FreeRTOS.

pavanshah2001, 2018-06-02:
@Noggin01 , Got it, thanks for the reply!!

aggarg, 2018-06-06:
As ESP32 is now supported, I am closing this. Let us know if you need anything else.

Thanks.

closed 2018-06-06 by aggarg