aws/amazon-freertos #5: Is plain freertos dead?

marekr, 2017-12-02:
Is it dead? Will we be stuck with Amazon pushing IoT libraries the majority of embedded projects don't use?

kbumsik, 2017-12-03:
You still be able to the plain [FreeRTOS v10 from its hompage](https://www.freertos.org/), without AWS-related stuff though.

ourairquality, 2017-12-04:
Where will FreeRTOS development continue? In the svn repo, or here on github?

Would this repo be well described as cherry picked and re-organised FreeRTOS code bundled with other code for AWS?

feilipu, 2017-12-07:
Just downloaded FreeRTOS-10 from the normal Sourceforge location, and converted it to being an [Arduino Library](https://github.com/feilipu/Arduino_FreeRTOS_Library), as is my usual process.

I can't see Richard Barry forgetting he spent 15 years building FreeRTOS as an open source project.

adjervis, 2017-12-07:
The FreeRTOS.org website continues to be the home of the FreeRTOS kernel (https://www.freertos.org/FreeRTOS-V10.html), and AWS will continue to develop and support the FreeRTOS kernel – as evidenced by FreeRTOS V10, which is Amazon’s first release, and contains new features as well as simplified licensing.

closed 2018-05-07 by drbartling