aws/amazon-freertos #6: Support for Raspberry Pi

bradspry, 2017-12-04:
Just look at those beautiful I2C ports waiting to send their data to AWS IoT:

[image]

[image]

[I2C shield source](https://store.ncd.io/?fwp_product_type=accessories&fwp_platform=raspberry-pi)

dave2, 2017-12-06:
Given the Pi is just Linux on a MPU-class SOC, you can just use Python or any other larger implementation straight on the Pi, eg https://github.com/aws/aws-iot-device-sdk-python

FreeRTOS is for MCU-class systems with limited resources.

adjervis, 2017-12-07:
Amazon FreeRTOS does not currently support Raspberry Pis. As pointed out by an earlier post, the OS is being targeted to more resource constrained devices than the Raspberry Pi. However 3rd party code is available at : https://github.com/jameswalmsley/RaspberryPi-FreeRTOS.

bradspry, 2017-12-10:
It surely would be nice to have a lightweight alternative to Linux, optimized out of the box for AWS IoT.

drbartling, 2018-05-07:
As mentioned by @adjervis, Amazon FreeRTOS is targeted at smaller μCU level devices and does not currently support Raspberry-PI type devices. I will close this issue for now. Let us know if you need any other information.

closed 2018-05-07 by drbartling