aws/amazon-freertos #7: Porting to other platforms guide

pfried, 2017-12-08:
I know there will never be enough time to support all devices, but please add a porting guide to other platforms, so other developers can do it.

feilipu, 2017-12-08:
It may not be obvious, but this amazon-freertos is only a very small increment from [FreeRTOS+TCP](https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/index.html).

Richard has already ported FreeRTOS to over 50 platforms,and his [porting guide](https://www.freertos.org/a00090.html) and [list of existing ports](https://www.freertos.org/RTOS_ports.html) is available on his [FreeRTOS.org](https://www.freertos.org/) web site.

It would be worth reading all the information in detail there, as a first point of call.

pfried, 2017-12-08:
Okay, misunderstanding here: I do not need a porting guide of freertos itself, but the library construct as it is presented here. We do have a custom stack containing of freerots, mbedtls and aws-device-sdk.

But of course we would consider aws freertos if we knew what is necessary to use it for our platform. Right now it lists 3 supported platforms, but of course freertos itself supports way more.

pfried, 2017-12-08:
Okay, I found a porting section in the manual on [page 45](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ug.pdf#what-is-amazon-freertos)

closed 2017-12-08 by pfried