aws/amazon-freertos #11: Long wait on MQTT Connect

sblancodiez, 2017-12-28:
Hello,

I am playing with Amazon FreeRTOS and when I connect to the broker I observe a delay equal to the timeout value (something like 12 or 14 seconds) before the ACK is processed.

I think the problem is that after sending the connection message the prvManageConnections function does not get anything in the socket read and then sets the next timeout to the full timeout. I think it should check the socket often before waiting for the full time.

Maybe it is a naive fix but I changed in MQTT_Periodic the following:

/* If this Tx operation has not timed out yet, update when
 * the next earliest timeout will happen. */
 ulNextTimeoutTicks = mqttMIN( ulNextTimeoutTicks, mqttbufferGET_PACKET_TIMEOUT_TICKS( xBuffer ) );

to the following (including the FreeRTOS header):

/* If this Tx operation has not timed out yet, update when
 * the next earliest timeout will happen. */
 ulNextTimeoutTicks = mqttMIN( ulNextTimeoutTicks, 300 / portTICK_PERIOD_MS );

This way, it will check every 300ms for data in the socket instead of waiting.

Is this correct?

RichardBarry, 2017-12-28:
Hi - thanks for the information.  We will have a look and report back.  Which chip/board are you using?

sblancodiez, 2017-12-29:
Hi,

Custom board with an Infineon XMC4500, ethernet connection.

alexa-noxon, 2017-12-30:
Thanks for the update.  

Are you using v1.0.1 of aws_mqtt_agent.c/aws_mqtt_lib.c?  v1.0.1 is the most recent and was uploaded about a week and a half ago.   You can find the file version at the top of the file above the license agreement.

Have you ported configPRINTF?  If so, could you share the log outputs?

dcgaws, 2018-01-02:
sblancodiez, happy new year! The behavior you're describing also reminds me of a common gotcha that occurs when your device has an otherwise valid AWS IoT credential, but the associated policy in the AWS IoT "Thing" registry is incorrect or incomplete. The resulting behavior in that case is that the device will connect TCP, negotiate TLS, but then immediately be disconnected by the server. Essentially, the authentication succeeds, but the subsequent authorization check fails. 

If that sounds like possibly what you're observing, can you please review [this forum thread](https://forums.aws.amazon.com/thread.jspa?messageID=818875&#818875)? In summary, that thread describes how to use the AWS CLI to double-check "Thing" credential and policy configuration.

sblancodiez, 2018-01-12:
Hi,

Right now I can't run it for logging, I will do that when possible.

I used both 1.0.0 and 1.0.1 with the same results. The issue is that if when the MQTT task does the first non-blocking recv it does not get data, it sleeps the full timeout value before trying recv again. Since the first recv is called very fast after sending the connect command, and it has a long timeout, you get a very long and not needed wait.

I am not using AWS, I am testing against a Mosquitto broker in my local network. Also, I am not using the demo exactly as is; I am using TLS so obviously there is a parameter change in the setup part.

Best regards,

sblancodiez, 2018-01-15:
Yes, all that was already accounted for and it is working. The issue and possible solution is slow connection because how timeouts are managed. Since I did not write the library I just wanted to share that and ask if there is a negative consequence on the change I proposed, but without change it works although I get around 12 seconds of wait in the connect phase and with my change I get no appreciable wait (300ms as written).

aggarg, 2018-01-17:
In order to ensure that the MQTT agent wakes up frequently to check if any data is received on the connected socket, you should define the macro mqttconfigMQTT_TASK_MAX_BLOCK_TICKS (in aws_mqtt_agent_config.h) to a smaller value. Note that the frequent wake up of the MQTT agent to check the received data on the connected socket can be avoided if the board has a capability to inform the MCU (probably some interrupt) about the data received on a connected socket.

closed 2019-05-02 by dcgaws