aws/amazon-freertos #13: Support for ATMega328 or ATMega2560

rajibgupta, 2018-01-06:
Being one of the popular and widely used controllers around the globe, it deserves to be supported by amazon-freertors :)

RichardBarry, 2018-01-06:
You can find some third party updated kernel ports for a wider variety of AVR devices in the Atmel forum on the FreeRTOS interactive site.  https://interactive.freertos.org. Hope that helps.

rajibgupta, 2018-01-11:
Hi @RichardBarry 

Thanks for the very useful info. Now I am trying to use the free-rtos with the atmel controller. It might be tough to implement the AWS-IOT part though.

Thanks

feilipu, 2018-01-12:
It will be tough with an ATmega328p as it is very small, but with an ATmega2560 there shouldn't be too much trouble, provided you're using a hardware driven Ethernet or Wifi device.

If you choose a Wiznet 5500, 5200, or 5100 device, then I've already collected the code you'll need into one [place](https://github.com/feilipu/avrfreertos/tree/master/freeRTOS10xx/lib_iinchip).

You'll need to connect the MQTT implementation etc from this repository on top of either IPv4 (easiest), or IPv6 (harder but doable with [uIP ](https://github.com/feilipu/avrfreertos/tree/master/freeRTOS10xx/lib-uIP)and Wiznet MACRAW mode).

It would be easier to use the FreeRTOS+Plus+TCP layer, and just implement a network layer interface for the W5x00 in (in normal or IPRAW mode), copied from these [examples](https://github.com/aws/amazon-freertos/tree/master/lib/FreeRTOS-Plus-TCP/source/portable/NetworkInterface).

Good luck.

shchers, 2018-05-14:
Hi @rajibgupta ,

I prepared port for ATMega328P (used in Arduino Nano and its clones).
Please use - https://github.com/aws/amazon-freertos/pull/28
Hope it will be merged to mainline.

With best regards,
Sergey

mradulan, 2018-08-13:
@shchers , There are some follow up questions on the pull request #28 .

mradulan, 2019-01-14:
```cpp 
printf()
```cpp

mradulan, 2019-02-11:
closing this issue. we have added it to our prioritization list.

closed 2019-02-11 by mradulan

attie44, 2019-09-19:
> Being one of the popular and widely used controllers around the globe, it deserves to be supported by amazon-freertors :)

Hi Rajib, have you been able to use gprs to connect to AWS?