aws/amazon-freertos #17: Unexpected socket close using Amazon FreeRTOS demo on NXP LPC54018 board

axelpeterwolf, 2018-02-13:
I’m trying to run the Amazon FreeRTOS demo on the NXP LPC54018 board, using SEGGER Embedded Studio.

https://www.segger.com/aws/
https://www.segger.com/downloads/eval/amazon-freertos-lpc54018.zip

Everything looks good and I’m getting all the way to the end, and then the socket gets closed unexpectedly…

As a result, the Hello World messages can't get published.
See Debug Terminal output below.

Any ideas why that would be?
Thanks!
 
 
Starting key provisioning...
Write root certificate...
Write device private key...
Write device certificate...
Key provisioning done...
Starting WiFi...
WiFi module initialized.
WiFi connected to AP cerano359.
IP Address acquired 10.0.0.30
Creating MQTT Echo Task...
MQTT echo attempting to connect to a2raen285ovswb.iot.us-east-2.amazonaws.com.
Sending command to MQTT task.
Received message 10000 from queue.
Looked up a2raen285ovswb.iot.us-east-2.amazonaws.com as 18.218.105.213
MQTT Connect was accepted. Connection established.
Notifying task.
Command sent to MQTT task passed.
MQTT echo connected.
MQTT echo test echoing task created.
Sending command to MQTT task.
Received message 20000 from queue.
MQTT Subscribe was accepted. Subscribed.
Notifying task.
Command sent to MQTT task passed.
MQTT Echo demo subscribed to freertos/demos/echo
Sending command to MQTT task.
Received message 30000 from queue.
**About to close socket.
Socket closed.**
Notifying task.
**Command sent to MQTT task failed.
ERROR:  Echo failed to publish 'Hello World 0'**
Sending command to MQTT task.
Received message 40000 from queue.
**MQTT_Publish failed!**
Notifying task.
**Command sent to MQTT task failed.
ERROR:  Echo failed to publish 'Hello World 1'**
Sending command to MQTT task.
Received message 50000 from queue.
**MQTT_Publish failed!**
Notifying task.
...
...

alexa-noxon, 2018-02-13:
I have seen this same sequence of error messages when the thing policy being used doesn’t have the permissions for the thing to publish.

If you go to the policy associated with the communication certificates you are using (In the IoT Console, Click Manage, then Things, then select the thing who’s credentials are in your project.  Select Security from the side bar, click on the certificate ID you are using for your project, and then select Policies.  Double check that the your policy allows you to publish to the topic.  

    {
      "Effect": "Allow",
      "Action": "iot:Publish",
      "Resource": "arn:aws:iot:ap-southeast-2:123456789101:*"
    }
You need something like above, where ap-southeast-2 is your region, and the 12345… number is your account number.

Does this fix the problem?

axelpeterwolf, 2018-02-13:
I did have the statement to publish in my policy, but just in that statement I missed replacing the placeholder with my actual account number!
I updated the policy and everything is working now!
Thanks!!  :-)

closed 2018-02-13 by axelpeterwolf