aws/amazon-freertos #18: Reconnect (the Paho library style)

stefanoco, 2018-02-15:
The Eclipse Paho MQTT library provides a method `mosquitto_reconnect()` useful in order to restart a broken connection to a broker. Is there anything similar here? What's the suggested policy?

alexa-noxon, 2018-02-15:
At this time a reconnect function is not provided, and applications must call connect again with the same connection parameters.  
Thank you for the input and we will consider support for automatic reconnect functionality in the future.

stefanoco, 2018-02-16:
Would you consider pull requests on this or any other development?

alexa-noxon, 2018-02-16:
We are still working on our pull request process and guidelines.  We welcome contributions to the project, but the decisions and criteria on whether each change will be merged will be shared in a forthcoming update.

xiongyu0523, 2018-11-14:
Any process on auto reconnect?

stefanoco, 2018-11-14:
can't see any contribution policy so far so didn't invest on this

lt72, 2019-02-07:
We are re-writing MQTT to be transport independent in the context of supporting BLE in Amazon FreeRTOS. We did not implement auto-reconnect yet, but it is definitely on our radar. Please check the [ble beta branch](https://github.com/aws/amazon-freertos/tree/feature/ble-beta) in this repo and/or the [v4_beta](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta) branch in C-SDK for the latest updates. It is a work in progress, but we'd like your feedback! 
P.S. the C-SDK is slightly ahead...

stefanoco, 2019-02-08:
@lt72 we may have a chance of testing this new branch in the next few weeks, thanks for commenting

dcgaws, 2019-08-16:
@stefanoco and @xiongyu0523, hello, our new MQTT library interface can be found in [iot_mqtt.h](https://github.com/aws/amazon-freertos/blob/master/libraries/c_sdk/standard/mqtt/include/iot_mqtt.h). While we have not implemented reconnect, we do have an [official contribution policy](https://github.com/aws/amazon-freertos/blob/master/CONTRIBUTING.md) and welcome pull requests.

closed 2019-08-16 by dcgaws