aws/amazon-freertos #19: Connection to MQTT fails as ping response is not received

shwetachauhan, 2018-02-15:
Hi,

In our set up we are observing that as ping response is not received, the connection to MQTT server fails with MQTTTimeout error, does application needs to reestablish the connection when this error happens. And why does the ping response stop coming from server?

alexa-noxon, 2018-02-19:
(1)	Yes, at this time the application does need to reestablish the connection after a connection error.  
(2)	Note that ping/ICMP is blocked by the AWS gateway (so if you tried ‘ping YOUR_ENDPOINT’, you would not expect a response due to security settings ).  Is this what you are observing?
(3) Please take a look at prvSetupConnection in aws_mqtt_agent.c to debug where the connection is failing. In other words, is it failing in the TCP connect, the TLS negotation, or the MQTT connection?

Are you using the Windows Simulator or a device?  If it is not a credential/policy issue, does the problem go away if you increase value of mqttconfigKEEP_ALIVE_TIMEOUT_TICKS (located in aws_mqtt_agent_config.h) for greater tolerance of network conditions?  

To debug further, could you share your configPRINTF outputs?  Note that there is a debug log config in aws_mqtt_config.h (mqttconfigENABLE_DEBUG_LOGS) for more information.    You could also try to enable logging (https://docs.aws.amazon.com/iot/latest/developerguide/monitoring_overview.html#iot-using-cloudtrail) to view API calls.

shwetachauhan, 2018-02-19:
I am running this on LPC54018 based device,the issue happens when the device has been running for some time (15 minutes to 2 hours) & sending ping request and receiving the ping response continuously as keep alive mechanism. Suddenly after some time the device stop receiving the ping responses and MQQT connection timeout. So, i suspect it is related to security settings. 

I will try to increase the mqttconfigKEEP_ALIVE_TIMEOUT_TICKS and see if it improves the condition.

mradulan, 2019-01-07:
Hi @shwetachauhan , did the increase in mqttconfigKEEP_ALIVE_TIMEOUT_TICKS improve the condition?

abhidixi11, 2019-01-15:
I'm closing this issue, since it looks like almost an year old and there is no response from the poster. Please open another issue if the problem still exists.

closed 2019-01-15 by abhidixi11

shwetachauhan, 2021-09-16:
> Hi @shwetachauhan , did the increase in mqttconfigKEEP_ALIVE_TIMEOUT_TICKS improve the condition?

Yes it did , thanks for the suggestion