aws/amazon-freertos #20: MQTT echo test could not connect to broker

oaelectronics, 2018-02-27:
Hi,

I am trying to connect my PIC32MZ EF curiosity board. Unfortunatly I am not able to know what is going wrong > 

Firmware ver     : 4.2.1
Min driver ver : 4.2.1
Curr driver ver: 4.2.1
WILC1000: Initialization successful!

Start Wi-Fi Connection...
Wi-Fi Connected
2 7991 [IP-task] vDHCPProcess: offer ac140a08ip
3 13282 [IP-task] vDHCPProcess: offer ac140a08ip
4 13283 [IP-task] Creating MQTT Echo Task...
5 13283 [IP-task]

IP Address: 172.20.10.8
6 13283 [IP-task] Subnet Mask: xxx.xxx.xxx.xxx
7 13283 [IP-task] Gateway Address: xxx.xxx.xxx.xxx
8 13283 [IP-task] DNS Server Address: xxx.xxx.xxx.xxx

9 13285 [MQTTEcho] MQTT echo attempting to connect to <endpoint>.iot.eu-central-1.amazonaws.com.
10 13285 [MQTTEcho] Sending command to MQTT task.
11 13285 [MQTT] Received message 10000 from queue.
12 13543 [IP-task] Socket sending wakeup to MQTT task.
13 17602 [MQTT] Received message 0 from queue.
14 17689 [IP-task] Socket sending wakeup to MQTT task.
15 17689 [MQTT] Received message 0 from queue.
16 17701 [IP-task] Socket sending wakeup to MQTT task.
17 17701 [MQTT] Received message 0 from queue.
18 17703 [MQTT] About to close socket.
19 17802 [IP-task] Socket sending wakeup to MQTT task.
20 17865 [MQTT] Socket closed.
21 17865 [MQTT] Notifying task.
22 17865 [MQTT] Received message 0 from queue.
23 17866 [MQTTEcho] Command sent to MQTT task failed.
24 17866 [MQTTEcho] ERROR:  MQTT echo failed to connect.
25 17866 [MQTTEcho] MQTT echo test could not connect to broker.
26 17866 [MQTTEcho] Sending command to MQTT task.
27 17866 [MQTT] Received message 20000 from queue.
28 17866 [MQTT] Notifying task.
29 17867 [MQTTEcho] Command sent to MQTT task passed.
30 17867 [MQTTEcho] MQTT echo demo finished.

Somebody can help me with this issue ?

sarenameas, 2018-02-28:
Hello,
 
I am able to reproduce your issue when I do not have a policy or the correct policy attached to the certificate that is attached to my Thing.
 
Please verify that your AWS IoT Thing has the correct policy. 
On the **AWS IoT console**, in the left navigation panel go to **Manage > Things**
Click on your Thing. In your Thing’s information page on the left navigation panel click on  **Security **.
Click on the Certificate attached your Thing.
In the Certificate’s information page, in the left navigation panel, click on  **Policies **.
Please make sure there is a policy attached to this certificate. If there are none you can click  **Actions ** in the upper right, then **Attach policy**.
 
If you do not have a policy you will have to make a policy. You can do this under the **Secure > Policies** section of the AWS IoT console home page.
 
If you do have a policy or are making a new policy make sure you have  iot:Publish, iot:Subscribe, iot:Connect, iot:Receive actions allowed for all resources. To ensure you have all IoT actions enabled for your thing your policy document can allow all iot actions and look like:
 
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": "arn:aws:iot:us-west-2:123123123123:topic/replaceWithATopic"
    }
  ]
}
 
Replace us-west-2 with your region.
Replace 123123123123 with your AWS account number.

oaelectronics, 2018-03-01:
Hi Sarena,

 

Thank you for this quick answer.

Here is the policy attached to my thing:

{

  "Version": "2012-10-17",

  "Statement": [

    {

      "Effect": "Allow",

      "Action": [

        "iot:*",

        "iot:Publish",

        "iot:Subscribe",

        "iot:Connect",

        "iot:Receive",

        "iot:UpdateThingShadow",

        "iot:GetThingShadow",

        "iot:DeleteThingShadow"

      ],

      "Resource": "arn:aws:iot:eu-central-1:839157702747:thing/curiosity"

    }

  ]

}

The only difference I can see here is in the resource . You wrote “ topic “ instead of a thing . What is the topic should I write ?

Do you think the error could be on this difference ?

 

Thank you,

Regards,

Othman AYOUR

sarenameas, 2018-03-01:
Hello,

The “topic” and “replaceWithATopic” part should be replaced with the MQTT Echo demo topic name. 
If you want to allow all IoT actions (“iot:*”) on all MQTT topics:

{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": [
        "arn:aws:iot:<aws-region>:<aws-account-id>:*"
      ]
    }
  ]
}

I recommend you create a new policy to attach to your certificate, perhaps you can name it “MQTTEchoDemoPolicy”, with the following more granular policy:

{
    "Version": "2012-10-17",
    "Statement": [
    {
        "Effect": "Allow",
        "Action": "iot:Connect",
        "Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:*"
    },
    {
         "Effect": "Allow",
         "Action": "iot:Publish",
         "Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:topic/freertos/demos/echo"
    },
    {
         "Effect": "Allow",
         "Action": "iot:Subscribe",
         "Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:topicfilter/freertos/demos/echo"
     },
     {
         "Effect": "Allow",
         "Action": "iot:Receive",
         "Resource": "arn:aws:iot:<aws-region>:<aws-account-id>:topic/freertos/demos/echo"
     }  
     ]
}

The more granular policy is recommended for better security and because many specific topics map to each of the IoT actions that are need for the MQTT Echo Demo.
Please see https://docs.aws.amazon.com/freertos/latest/userguide/freertos-getting-started.html for more information about AWS Account and Permissions for getting started with Amazon FreeRTOS.

oaelectronics, 2018-03-02:
Thank you very much it worked when I modified the IoT subscribe action : topicfilter instead of topic.
Regards.

closed 2018-03-02 by oaelectronics

A04236718, 2019-03-11:
Hello

I'm also currently facing the same issue, I've applied the changes mentioned in the thread, but it's still not able to connect fully. Some context, I'm using Node-Red which will retrieve data from a sensor, but I want to be able to store that data on the AWS cloud storage facility. I'm unable to do so without having the IoT connection. Please let know if any other information is needed.

Thanks in advance

[info] [mqtt-broker:new] Connection failed to broker: mqtts://a3dv5lazqbff2z-ats.iot.us-east-2.amazonaws.com:8883

Thanks