aws/amazon-freertos #22: 2G Modules

ubuntudeveloper, 2018-03-05:
Hi ,
the current code sends data over wifi. is there a port available for 2G M2M modules like via AT commands.
If not, where can one start?

prasadvya, 2018-03-09:
Thank you for your inquiry. The ports available now does do not include a library for connecting to external 2G M2M modules but a custom driver based on AT commands can be developed and built with the project.

bandinigo, 2018-06-12:
Does AWS plan to build such a custom driver? And if yes, by when will it be available?  

Alternatively, do you know of any existing projects? 

We would like to use a [4G modem](https://www.gemalto.com/m2m/solutions/modules-terminals/industrial/els61) via AT commands...

dcgaws, 2018-06-12:
bandinigo, I recommend that you check with the Amazon FreeRTOS hardware partners. The latest list is here, https://aws.amazon.com/freertos/partners/. 

I'm not aware of an M2M driver that has been ported to Amazon FreeRTOS. That is, an M2M solution that uses our socket API, for example. However, I have seen M2M solutions from the above vendors that support FreeRTOS. In other words, the driver works with the kernel and the particular hardware. If you wanted to port it to our newer socket abstraction layer, you could write a wrapper.

sudarshangb, 2018-06-19:
Hello dcgaws, 
Thank you for the suggestion. I want to use the SIM800L GSM module with ESP32 with Amazon FreeRTOS. I am relatively new to this. Can you please explain in a bit detail about how I can use the GSM module with AFR? What exactly should the wrapper do?

dcgaws, 2018-06-19:
Hello, the Amazon FreeRTOS MQTT client (agent) uses our "secure sockets" interface (basically looks like Berkeley sockets with TLS support). If you have an alternative implementation of TLS and TCP/IP that uses GSM for media, you can integrate it with our MQTT agent at the secure sockets layer. Please see the header file aws_secure_sockets.h. 

However, some of our reference ports have preferred to use our secure sockets implementation (i.e., including the mbedTLS and FreeRTOS+TCP libraries) and instead just replace the media layer. If you think that approach might work for you, check out the subdirectories of lib\FreeRTOS-Plus-TCP\source\portable\NetworkInterface.

sfakir, 2018-10-31:
@sudarshangb I would also be interessted. Did you solve this problem?

mradulan, 2019-02-11:
closing this issue. we have added it to our prioritization list.

closed 2019-02-11 by mradulan

GeorgeGkinis, 2019-06-25:
As far as I understand **lwip** supports **pppos**. The problem is that **lwip** is not currently supported for **ESP32** which I intent to use.
Being relatively new to awsFREErtos I am unable to find a way to connect to awsCore using a modem (SIM7000). 
Using WiFi works perfectly, but I have no clue where to start to be able to use a cellular modem.

There are examples using **ESP-IDF** + **MQTT** + **lwip** + **pppos** but the **TCP/IP** stack used is **lwip**.

Currently there is no support for **lwip** for **esp32** as stated [here](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html).

My case could be resolved by porting the (custom) **lwip** stack from **ESP-IDF** and providing an example would also be greatly appreciated.

Has anyone managed to get the **MQTT** + **modem** + **ESP32** -> **awsIoTCore** working?
Please let me know, I have been struggling to find a solution for some days now... :/

mnelson920, 2019-07-24:
Has anyone got this working? Interested in using Amazon-FreeRTOS + ESP32 + U-blox SARA-R4 cell modem.

nickfaughey, 2019-07-25:
I'm currently working on getting this implemented (Amazon-FreeRTOS + STM32 + SIM800) by swapping out the HAL layer. I'll share when there's some tangible progress - it's still very early stages.

tanmoysen, 2019-07-25:
Hi mnelson920, we are exploring this and would be interested in a PR in this area. We would also like to discuss your project if you're open to that - you can reach out to freertos-qual at amazon dot com.

mnelson920, 2019-07-25:
Hi @tanmoysen, just sent an email to that address.

bszekely, 2019-08-02:
Hi @nickfaughey, please share your progress as I'm also planning to integrate STM32 + SIM800 with AWS IOT. I'm highly interested in if/how you were able to build this.

ZakiGit13, 2019-11-10:
hello

i am trying also to make this working using a SIM7600 ans a PIC24 controller.

if any update , it will be great.

maheshwargm, 2020-01-16:
How to send certificates using sim808 to AW.

alfriedrich, 2020-02-07:
Does anyone know of any progress related to this issue or any open source package that can be used?

ahmedwahdan, 2020-05-05:
any update on this ticket?

nickfaughey, 2020-05-05:
I've gotten this working with a SIM800. Source is proprietary, but I'll try to explain the approach and share some useful bits; sorry if it's vague. Thankfully, this framework is very discretely layered with clear interfaces between the different subsections, so all of the big amazon-freertos changes were limited to one file. For what it's worth, I did this swap from the default [STM32 IoT board demo](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_st.html), but the process should be similar for others.

I'm using MaJerle's awesome [SIM800 library](https://github.com/MaJerle/gsm-at-lib), which natively supports abstract driver hardware and provided a demo for STM32. It should be possible to support other microcontrollers, but that's out of scope for this topic; take it up in that repo.

**`aws_secure_sockets.c`** 
This is the file that handles the interface between the upper application layers and the low-level hardware abstractions. Mine was customized for my WiFi board, so I basically gutted the existing file and swapped out the WiFi functionality for GSM functionality.

The `SOCKETS_*` functions are the main entrypoints, called by upper layers. For example, `SOCKETS_Connect` has this line:
```
gsmr_t ret = gsm_conn_start(&xGsmConn, GSM_CONN_TYPE_TCP, str_ip_address, SOCKETS_ntohs(pxAddress->usPort), NULL, &conn_callback, 1);
```
That's where I start a SIM800 connection, and specify the new `conn_callback` function as the target for incoming data. It receives incoming bytes from the GSM library and stores them in a local ring buffer:

```
static gsmr_t conn_callback(gsm_evt_t* evt)
{
	// event from GSM connection

    switch (gsm_evt_get_type(evt)) {
    	case GSM_EVT_CONN_RECV: {
    		// the GSM connection has received data; we should copy it to the local buffer so it can continue to receive more

			gsm_pbuf_p buf = gsm_evt_conn_recv_get_buff(evt);

			size_t buff_len = 0, buff_offset = 0;
			const uint8_t * d;

			// because gsm lib dynamically allocates memory, it might not be contiguous and this function needs to be called many times to get the whole thing
			do {
		        buff_offset += buff_len;                					/* Calculate new offset of buffer */
				d = gsm_pbuf_get_linear_addr(buf, buff_offset, &buff_len);  /* Get address pointer */

				// copy whatever contiguous bytes we have here
				uint16_t i = 0;
				while (d != NULL && i < buff_len) {
					uint8_t data = d[i++];
					ring_buf_write(data);
				}
			} while (buff_len > 0);

			break;
		}
    }

	return gsmOK;
}
```

`prvNetworkRecv(...)` is called by the socket layer, and just reads out of that ring buffer:
```
...
// read the max of available and buffer length bytes from the local ring buffer
uint16_t availableBytes = ring_buf_count();
uint16_t bytesToRead = availableBytes;
if (xReceiveBufferLength < availableBytes) {
	bytesToRead = xReceiveBufferLength;
}

            /* Receive the data. */
for (uint16_t i = 0; i < bytesToRead; i++) {
	ring_buf_get((uint8_t *) pucReceiveBuffer + i);
}

// assume we successfully copied all from ring buffer
usReceivedBytes = bytesToRead;

            /* Return the semaphore. */
            ( void ) xSemaphoreGive( xGsmModule.xSemaphoreHandle );

xRetVal = (BaseType_t) usReceivedBytes;
if (usReceivedBytes > 0) {
	// success, return the number of bytes read
	break;
} else {
	// no bytes read. if we've timed out, return back
	if (xTaskGetTickCount() - xTimeOnEntering > 1000) {
		break;
	} else {
		// there still might be bytes coming in. delay and try again
		vTaskDelay( stsecuresocketsFIVE_MILLISECONDS );
	}
}
...
```

`prvNetworkSend(...)` is considerably simpler because we just pass the outgoing data to the GSM layer:
```
...
xGsmResult = gsm_conn_send(
    xGsmConn,		// handle to the (hopefully) active connection
    pucData, 		// payload
    xDataLength,	// length of payload
    &xBytesSent, 	// output number of bytes actually sent
    1);			// blocking
...
```

This is the bulk of the replacements. I also swapped out `aws_wifi.c` with my own `aws_gsm.c` and replaced its references in the upper layers for handling the initialization and control flow, but that architecture is up to you. Ex. `WIFI_On()` -> `GSM_On()`, `WIFI_ConnectAP()` -> `GSM_ConnectTower()`, etc.

Probably out of scope for this, but you'll also want to implement `SOCKETS_GetHostByName()` to turn your alphabetic MQTT endpoint into an IP for the GSM connection. AWS uses a load balancer, so you can't count on a hard-coded IP to always resolve. The GSM library didn't build this command (`+CDNSGIP=<domain.tld>`) into the library, so I implemented it alongside the other commands.

Happy to try to answer questions here, but unfortunately my code is a bit tangled with proprietary stuff, and abstracting a simple demo for open source release isn't on the roadmap right now.

genotix, 2021-01-20:
> I've gotten this working with a SIM800. Source is proprietary, but I'll try to explain the approach and share some useful bits; sorry if it's vague. Thankfully, this framework is very discretely layered with clear interfaces between the different subsections, so all of the big amazon-freertos changes were limited to one file. For what it's worth, I did this swap from the default [STM32 IoT board demo](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_st.html), but the process should be similar for others.
> 
> I'm using MaJerle's awesome [SIM800 library](https://github.com/MaJerle/gsm-at-lib), which natively supports abstract driver hardware and provided a demo for STM32. It should be possible to support other microcontrollers, but that's out of scope for this topic; take it up in that repo.
> 
> **`aws_secure_sockets.c`**
> This is the file that handles the interface between the upper application layers and the low-level hardware abstractions. Mine was customized for my WiFi board, so I basically gutted the existing file and swapped out the WiFi functionality for GSM functionality.
> 
> The `SOCKETS_*` functions are the main entrypoints, called by upper layers. For example, `SOCKETS_Connect` has this line:
> 
> ```c
> gsmr_t ret = gsm_conn_start(&xGsmConn, GSM_CONN_TYPE_TCP, str_ip_address, SOCKETS_ntohs(pxAddress->usPort), NULL, &conn_callback, 1);
> ```
> 
> That's where I start a SIM800 connection, and specify the new `conn_callback` function as the target for incoming data. It receives incoming bytes from the GSM library and stores them in a local ring buffer:
> 
> ```c
> static gsmr_t conn_callback(gsm_evt_t* evt)
> {
> 	// event from GSM connection
> 
>     switch (gsm_evt_get_type(evt)) {
>     	case GSM_EVT_CONN_RECV: {
>     		// the GSM connection has received data; we should copy it to the local buffer so it can continue to receive more
> 
> 			gsm_pbuf_p buf = gsm_evt_conn_recv_get_buff(evt);
> 
> 			size_t buff_len = 0, buff_offset = 0;
> 			const uint8_t * d;
> 
> 			// because gsm lib dynamically allocates memory, it might not be contiguous and this function needs to be called many times to get the whole thing
> 			do {
> 		        buff_offset += buff_len;                					/* Calculate new offset of buffer */
> 				d = gsm_pbuf_get_linear_addr(buf, buff_offset, &buff_len);  /* Get address pointer */
> 
> 				// copy whatever contiguous bytes we have here
> 				uint16_t i = 0;
> 				while (d != NULL && i < buff_len) {
> 					uint8_t data = d[i++];
> 					ring_buf_write(data);
> 				}
> 			} while (buff_len > 0);
> 
> 			break;
> 		}
>     }
> 
> 	return gsmOK;
> }
> ```
> 
> `prvNetworkRecv(...)` is called by the socket layer, and just reads out of that ring buffer:
> 
> ```c
> ...
> // read the max of available and buffer length bytes from the local ring buffer
> uint16_t availableBytes = ring_buf_count();
> uint16_t bytesToRead = availableBytes;
> if (xReceiveBufferLength < availableBytes) {
> 	bytesToRead = xReceiveBufferLength;
> }
> 
>             /* Receive the data. */
> for (uint16_t i = 0; i < bytesToRead; i++) {
> 	ring_buf_get((uint8_t *) pucReceiveBuffer + i);
> }
> 
> // assume we successfully copied all from ring buffer
> usReceivedBytes = bytesToRead;
> 
>             /* Return the semaphore. */
>             ( void ) xSemaphoreGive( xGsmModule.xSemaphoreHandle );
> 
> xRetVal = (BaseType_t) usReceivedBytes;
> if (usReceivedBytes > 0) {
> 	// success, return the number of bytes read
> 	break;
> } else {
> 	// no bytes read. if we've timed out, return back
> 	if (xTaskGetTickCount() - xTimeOnEntering > 1000) {
> 		break;
> 	} else {
> 		// there still might be bytes coming in. delay and try again
> 		vTaskDelay( stsecuresocketsFIVE_MILLISECONDS );
> 	}
> }
> ...
> ```
> 
> `prvNetworkSend(...)` is considerably simpler because we just pass the outgoing data to the GSM layer:
> 
> ```c
> ...
> xGsmResult = gsm_conn_send(
>     xGsmConn,		// handle to the (hopefully) active connection
>     pucData, 		// payload
>     xDataLength,	// length of payload
>     &xBytesSent, 	// output number of bytes actually sent
>     1);			// blocking
> ...
> ```
> 
> This is the bulk of the replacements. I also swapped out `aws_wifi.c` with my own `aws_gsm.c` and replaced its references in the upper layers for handling the initialization and control flow, but that architecture is up to you. Ex. `WIFI_On()` -> `GSM_On()`, `WIFI_ConnectAP()` -> `GSM_ConnectTower()`, etc.
> 
> Probably out of scope for this, but you'll also want to implement `SOCKETS_GetHostByName()` to turn your alphabetic MQTT endpoint into an IP for the GSM connection. AWS uses a load balancer, so you can't count on a hard-coded IP to always resolve. The GSM library didn't build this command (`+CDNSGIP=<domain.tld>`) into the library, so I implemented it alongside the other commands.
> 
> Happy to try to answer questions here, but unfortunately my code is a bit tangled with proprietary stuff, and abstracting a simple demo for open source release isn't on the roadmap right now.

Hi Nick,

First of all let me say how happy I am that you got this working with AWS.
Is there some way to purchase the proprietary code that enables FreeRTOS + SIM800L towards AWS?