aws/amazon-freertos #23: is support for BLE transport planned ?

xnlcasad, 2018-03-12:
All is in the issue title; is amazon planning to add bluetooth low energy support to their version of freertos? Thanks for your comments.

sarenameas, 2018-03-13:
Thank you for your inquiry. As you already know, we do not currently support BLE. My team values your inquiry and we will use it in our consideration for prioritization on our support for BLE.

rzr, 2018-06-11:
Which other port do you suggest as fallback meanwhile it is available ? I am considering espressif one

aggarg, 2018-06-18:
Thank you for your inquiry. Amazon FreeRTOS does not support BLE as of now. Please choose a port best suitable for your application needs.

Your feedback is important to us as it helps us in prioritizing our work.

Thanks.

sudarshangb, 2018-07-06:
Hello,

I am planning to use BLE with Amazon FreeRTOS  (on ESP32) as soon as possible. Can we have an estimate about when it will be available?

Thanks!

tanmoysen, 2018-07-08:
Hi sudarshangb,
 
Thanks for your inquiry and interest. 
We currently don’t have a date for having BLE support on ESP32, but feedback such as this helps us prioritize. It would be appreciated if you could share some details on your BLE use case/requirements.
 
Best,
Tanmoy

dwlayton, 2018-08-31:
I have two use cases for BLE on a custom ESP32 module that  I'm working on:  1) provisioning of the ESP32 via a mobile app so that it can connect to a local wifi/AWS IoT and 2) connecting to other BLE sensors and devices that then pass data to AWS Iot via the ESP32.

dwlayton, 2018-08-31:
I was hoping that the Espressif BLE would work along with the Amazon FreeRTOS--and your response supports this view. Thanks!

refuhoo, 2018-09-04:
Hi David,

How do you plan to have Espressif BLE would work along with the Amazon FreeRTOS? If it works for you, do you mind sharing your solution? Thanks!

dwlayton, 2018-09-04:
My tentative plan is to use an app to provision the wifi settings (network name and password) on the Espressif device via BLE. However, I have not implemented this yet--

refuhoo, 2018-09-12:
Hi Tanmoy,

You said "While we don't have a standardized BLE interface for Amazon FreeRTOS yet, our partners such as Espressif have solutions for BLE, and our software will work with these solutions."

Can you give more explanation of how to do it?

thanks!

lt72, 2018-09-12:
Hi @refuhoo,

the Espressif [boards ](https://aws.amazon.com/about-aws/whats-new/2018/05/esp32-qualified-for-amazon-freertos/) qualified for Amazon FreeRTOS have BLE connectivity but this repo neither contains the necessary libraries from Espressif idf project, nor it contains a standardized BLE interface we are committed to support. You can find the source code for Espressif BLE libraries on github at [esp-idf](https://github.com/espressif/esp-idf/tree/master/components/bt). 

I can see two ways to leverage BLE on the Espressif boards: one can either import the necessary libraries into this code base, or one can use the FreeRTOS version in the Espressif code base. 
In the first approach, one should first import the BLE libraries (see link above) into this code base, make sure they build correctly and then code/test as usual. 
In the second approach, one would work directly with the version of FreeRTOS that is available in the esp-idf project itself. See [here](https://github.com/espressif/esp-idf/tree/master/components/freertos).

The first approach will require some preliminary steps, but will allow using all other Amazon FreeRTOS libraries (MQTT, Shadows, OTA, etc...). The risk is that one may find incompatibilities between the current esp-idf project sources and the sources in this repo. We do not guarantee in facts that esp-idf and Amazon FreeRTOS are aligned at all times. 
The second approach will only depend from the esp-idf project, and should allow immediate prototyping. The downside is that the Amazon FreeRTOSlibraries are not available in the esp-idf repo.

In the event that we will decide to add BLE support to this code base, we will first define a standardized interface and then implement it against the ESP BLE API and whatever other API is available with other qualified SoCs. A standard interface is needed so that any BLE application running on Amazon FreeRTOS can be ported across all qualified boards/SoCs/etc... 
One last thing to consider is that we did not commit on delivering BLE support yet, and therefore we did not define such abstraction as of today. Whatever solution one designs&code before such abstraction is available, it _may_ need re-coding once the abstraction is provided and implemented by our team.

lt72, 2018-11-27:
@refuhoo @dwlayton @sudarshangb @xnlcasad Hi all, yesterday we launched the Beta for BLE support for 3 targets: Espressif, Nordic and ST. 
Espressif and ST are the boards we already support today, although for ST you need to use a different BLE SoC than the one on-board. For Nordic, we do support the nRF52840-DK. Our current implementation, still Beta quality (!), supports MQTT-over-BLE using a proxy iOS or Android device. We also support Shadows as well as WiFi provisioning. Of course we support creating custom GATT services as well. Please check our [online docs](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-ble-library.html) for more details, as well as the [iOS](https://github.com/aws/amazon-freertos-ble-ios-sdk) and [Android](https://github.com/aws/amazon-freertos-ble-android-sdk) SDK repos, which also feature a sample application. 
Espressif has the more complete and well tested implementation so far, with Nordic an ST catching up quickly. ST does not support WiFi provisioning yet, but we are working on it and it  won't be long. Nordic does not have a WiFi SoC at all.

If you happen to give this feature a try, we are very interesting in your feedback and suggestions.

dwlayton, 2018-12-05:
@it72 thanks for the update on BLE support! I have a prototype pcb using the new Espressif Solo-1 module. My use case includes an Ionic app that uses the AWS Amplify framework that supports Cognito authentication and AWS iot pub/sub. I will use the new ble support for aws freertos  to provision the wifi connection on the module using the Ionic app with BLE (i.e., sends the wifi SSID and pw to a custom BLE service set up on the Solo-1). 
Once connected to AWS IoT, I was hoping to do JITP, however, looking ahead to a production environment with thousands of pcb units produced, flashing a unique x.509 certificate onto each device to allow JITP seems impracticable/expensive. Thus, I was interested in the solution provided by https://github.com/awslabs/aws-iot-certificate-vending-machine.

My question is--instead of having the Solo-1 make an https call to the API gateway and then to the Lambda provisioning function (as in the CVM example), couldn't I make a similar  https call from the Ionic app (using the device serial number read from the BLE service + cognito user id) and then send the acquired CVM certs to the Solo-1 via the BLE connection (using a custom BLE service)? One key advantage is that I could then attach the authenticated cognito user id (from the app) to the Thing shadow that the device publishes to.

lt72, 2019-02-07:
@dwlayton: In our BLE use case, the actual 'device', as far as the MQTT broker is concerned, is the proxy. So forwarding the Cognito identity to the device is not needed. 
That said, if you want to use BLE just for WiFi provisioning, and then do all MQTT transactions or WiFi, then it would make sense. 
Am I guessing right?

gordonwang0, 2019-05-08:
Hi everyone,

BLE support has been added for select boards on the release-1.5 branch.  I am closing this issue; please reopen if you have any additional questions.

closed 2019-05-08 by gordonwang0