aws/amazon-freertos #24: Unpatched upstream vulnerabilities in mbedtls libraries

chan0415, 2018-04-15:
There are several unpatched upstream vulnerabilities in mbedtls library. 

* **CVE-2018-9988**
Brief description: Buffer over-read in `ssl_parse_server_key_exchange()` that could cause a crash on invalid input.
upstream patches: https://github.com/ARMmbed/mbedtls/commit/027f84c69f4ef30c0693832a6c396ef19e563ca1, https://github.com/ARMmbed/mbedtls/commit/a1098f81c252b317ad34ea978aea2bc47760b215
NVD link: https://nvd.nist.gov/vuln/detail/CVE-2018-9988

* **CVE-2018-9989**
Brief description: Buffer over-read in `ssl_parse_server_psk_hint()` that could cause a crash on invalid input.
upstream patches: https://github.com/ARMmbed/mbedtls/commit/5224a7544c95552553e2e6be0b4a789956a6464e, https://github.com/ARMmbed/mbedtls/commit/740b218386083dc708ce98ccc94a63a95cd5629e
NVD link: https://nvd.nist.gov/vuln/detail/CVE-2018-9989

* **CVE-2018-0487**
Brief description: Allows remote attackers to execute arbitrary code or cause a denial of service (buffer overflow) via a crafted certificate chain that is mishandled during RSASSA-PSS signature verification within a TLS or DTLS session
upstream patches: https://github.com/ARMmbed/mbedtls/commit/28a0c727957990ac655cbe40c7eb20b7ef01167d
NVD link: https://nvd.nist.gov/vuln/detail/CVE-2018-0487
Advisory: https://tls.mbed.org/tech-updates/security-advisories/mbedtls-security-advisory-2018-01

* **CVE-2018-0488**
Brief description: When the truncated HMAC extension and CBC are used, allows remote attackers to execute arbitrary code or cause a denial of service (heap corruption) via a crafted application packet within a TLS or DTLS session
upstream patches: https://github.com/ARMmbed/mbedtls/commit/992b6872f3ca717282ae367749a47f006d337a87, https://github.com/ARMmbed/mbedtls/commit/464147cadc694379b7717afb7b517fe05cdb323f
NVD link: https://nvd.nist.gov/vuln/detail/CVE-2018-0488
Advisory: https://tls.mbed.org/tech-updates/security-advisories/mbedtls-security-advisory-2018-01

* **CVE-2017-18187**
Brief description:  There is a bounds-check bypass through an integer overflow in PSK identity parsing in the `ssl_parse_client_psk_identity()` function in `library/ssl_srv.c`.
upstream patches: https://github.com/ARMmbed/mbedtls/commit/83c9f495ffe70c7dd280b41fdfd4881485a3bc28
NVD link: https://nvd.nist.gov/vuln/detail/CVE-2017-18187

If possible, upgrade to the latest mbedtls 2.8.0 version

alexa-noxon, 2018-04-16:
Thank you for pointing this out.  Amazon FreeRTOS is currently using mbedTLS version 2.7 which is behind the latest version.  We are looking into upgrading versions.

drbartling, 2018-05-14:
Hello,

We have patched mbed TLS with the dot release of Amazon FreeRTOS 1.2.4.  The version of mbed TLS is no 2.8.

FOr further information, please see the change log at:  
https://github.com/aws/amazon-freertos/blob/v1.2.4/change_log.txt

closed 2018-05-14 by drbartling