aws/amazon-freertos #25: Update request pic32mzef MHC

povi2946, 2018-04-18:
Could we get an updated Harmony configuration file?

 amazon-freertos/demos/microchip/curiosity_pic32mzef/common/application_code/microchip_code/pic32mz_ef_curiosity.mhc 

Which includes newest wifi configurations.

Thanks

dcgaws, 2018-04-23:
povi2946, hi, you've got our latest. Can you please be more specific about what you'd like to see added?

povi2946, 2018-04-23:
Once mhc file is loaded any SPI and Wifi configurations are not in the configuration.

dcgaws, 2018-04-24:
Ah, sorry, I only looked at the diff of the file and didn't recognize the actual issue. The MHC files in our repo are from the initial ports from Harmony over to Amazon FreeRTOS. However, the Microchip Harmony Configurator doesn't recognize the Amazon FreeRTOS directory structure, and the MHC files in our repo are not being maintained. 

I regret the inconvenience that this is causing to customers that are accustomed to MHC. We're discussing potential longer-term solutions.

aggarg, 2018-06-06:
Thank you for the feedback. It is important as it helps in prioritizing our work. 

I am closing this issue for now. We will keep you updated if anything changes.

Thanks.

closed 2018-06-06 by aggarg

jadamroth, 2019-04-13:
Have there been any updates to this?

Reconfiguring this repo into a harmony compatible project to where it properly compiled was super annoying. I've seen several forums of people citing similar struggles to the ones mentioned above. 

Thanks