aws/amazon-freertos #26: PIC32 MAC address not working

povi2946, 2018-05-11:
as in file 
 amazon-freertos/demos/microchip/curiosity_pic32mzef/common/config_files/FreeRTOSConfig.h

Lines 208-216
/* PIC32 note: PIC32 processors have a factory programmed MAC address.
 * This address setting will override the factory programmed address.
 * Use all 0's to use the factory programmed address */

After setting all variables to 0 and running the demo device on my network appears with MAC address with all zeros instead of factory preprogrammed PIC32 MAC address.

ravibhagavandas, 2018-05-19:
Hello,
We tried to reproduce the scenario with Amazon FreeRTOS release 1.2.6 and saw that mac address was set **correctly** to the device mac address for the outgoing packets from the device. 
Please note that we overwrite the  default mac address when we connect to wifi AP here:
https://github.com/aws/amazon-freertos/blob/ce545b8faa691979cd6ce42ee492565f8c316309/lib/wifi/portable/microchip/curiosity_pic32mzef/aws_wifi.c#L307
Could you please point out how you verify that mac address for the device is set as 0 ? Also which version of FreeRTOS code are you using ?

mradulan, 2019-01-07:
Hi @povi2946 , did it solve your issue?

yuhui-zheng, 2019-01-31:
Since this issue has been inactive, closing. Please reach out again, if there's any question.

closed 2019-01-31 by yuhui-zheng