aws/amazon-freertos #27: Amazon FreeRTOS does not compile

RobertBerger, 2018-05-14:
Hi, I am trying to get this to work - initially to compile: 
*) NXP LPC54018 IoT Module 
*) mcuxpressoide-10.1.1_606.x86_64.deb.bin 
*) SDK_2.3.1_LPC54018-IoT-Module.zip 
*) freertos-1526159674761.zip *) 
on Ubuntu 16 64 bit.

I followed the instructions: 
https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_nxp.html 
But "Build all" has issues to find various header files. 

This is how a fresh install looks like: 
https://pastebin.com/3WfVsi3R 

> fatal error: wifi_common.h: No such file or directory #include "wifi_common.h" ^ compilation terminated. 

Please help 

Regards, 

Robert

RobertBerger, 2018-05-14:
I opened also a support case with Amazon: 

case 5061051661

RobertBerger, 2018-05-22:
Just wanted to add, that as described above I got it compile.

```
$ pwd
/home/rber/mcuxpresso/AmazonFreeRTOS
```
```
$ tree ../ -L 1
../
├── 01
└── AmazonFreeRTOS

2 directories, 0 files

```
```
$ `sed -i 's/lib\/third_party\/mcu_vendor\/nxp\/lpc54018/lib\/third_party\/mcu_vendor\/nxp\/LPC54018/g' ~/mcuxpresso/AmazonFreeRTOS/demos/nxp/lpc54018_iot_module/mcuxpresso/.project`
```

Thanks!

ravibhagavandas, 2018-05-23:
The bug has been fixed with the new release (1.2.6) of Amazon FreeRTOS. Please use the latest release and feel free to open an issue if you have further questions.

closed 2018-05-23 by aggarg