aws/amazon-freertos #30: FreeRTOS+TCP: Accept call does not return on connect if bPassQueued is not set (+ proposed fix)

forGGe, 2018-05-18:
Hello!

I'm using FreeRTOS + TCP/IP stack with K64 MCU from NXP.

I have written simple TCP echo server, and found that when `FreeRTOS_accept` call does not return when connection is established. Wireshark shows correct SYN/SYN-ACK/ACK procedure, but application does not receive any events.

I found out that application task does not receive `eSOCKET_ACCEPT` event on **listening socket**  from internal FreeRTOS tasks. 

Instead, `eSOCKET_CONNECT` is set for **child** socket that nobody waits for (`FreeRTOS_accept` waits for listening socket, as I mentioned):

https://github.com/aws/amazon-freertos/blob/e05348379795f4fd6650bc79a6d38f051eb5f5f5/lib/FreeRTOS-Plus-TCP/source/FreeRTOS_TCP_IP.c#L1502-L1513

### FreeRTOS+TCP/IP version

I'm using v2.0.1

### FreeRTOS configuration

```c
#define ipconfigTCP_MSS                             522
#define ipconfigTCP_TX_BUFFER_LENGTH                ( 2 * ipconfigTCP_MSS )
#define ipconfigTCP_RX_BUFFER_LENGTH                ( 2 * ipconfigTCP_MSS )
#define ipconfigNETWORK_MTU                         586
#define ipconfigNUM_NETWORK_BUFFER_DESCRIPTORS      4
#define ipconfigEVENT_QUEUE_LENGTH                  10
#define ipconfigUSE_LINKED_RX_MESSAGES              0
#define ipconfigBYTE_ORDER                          ( pdFREERTOS_LITTLE_ENDIAN )
#define ipconfigUSE_TCP_TIMESTAMPS                  0
#define ipconfigRAND32()                            rng_rand()
#define ipconfigUSE_NETWORK_EVENT_HOOK              1
#define ipconfigUSE_CALLBACKS                       1
#define ipconfigIP_TASK_STACK_SIZE_WORDS            256
#define ipconfigIP_TASK_PRIORITY                    3
#define ipconfigDHCP_SEND_DISCOVER_AFTER_AUTO_IP    0
#define ipconfigUSE_TCP_WIN                         0

```

### Test application code

```c
static void echo_demo(void)
{
    struct freertos_sockaddr xClient, xBindAddress;
    Socket_t xListeningSocket, xConnectedSocket;
    socklen_t xSize = sizeof( xClient );
    static const TickType_t xReceiveTimeOut = portMAX_DELAY;
    const BaseType_t xBacklog = 20;
    #define BUFFER_SIZE 512
    static char cRxedData[ BUFFER_SIZE ];
    BaseType_t xResult;

    /* Attempt to open the socket. */
    xListeningSocket = FreeRTOS_socket( FREERTOS_AF_INET,
                                        FREERTOS_SOCK_STREAM,  /* SOCK_STREAM for TCP. */
                                        FREERTOS_IPPROTO_TCP );

    /* Check the socket was created. */
    configASSERT( xListeningSocket != FREERTOS_INVALID_SOCKET );

    /* If FREERTOS_SO_RCVBUF or FREERTOS_SO_SNDBUF are to be used with
    FreeRTOS_setsockopt() to change the buffer sizes from their default then do
    it here!.  (see the FreeRTOS_setsockopt() documentation. */

    /* If ipconfigUSE_TCP_WIN is set to 1 and FREERTOS_SO_WIN_PROPERTIES is to
    be used with FreeRTOS_setsockopt() to change the sliding window size from
    its default then do it here! (see the FreeRTOS_setsockopt()
    documentation. */

    /* Set a time out so accept() will just wait for a connection. */
[... 140 lines cut ...]
 				{
+					/* Parent can (must?) be found in pxSocket->u.xTCP.pxPeerSocket,
+					otherwise setting bPassQueued for new sockets is meaningless */
 					xParent = pxSocket->u.xTCP.pxPeerSocket;
 					configASSERT( xParent != NULL );
 				}
@@ -3234,7 +3236,13 @@ struct freertos_sockaddr xAddress;
 	xAddress.sin_addr = *ipLOCAL_IP_ADDRESS_POINTER;
 	xAddress.sin_port = FreeRTOS_htons( pxSocket->usLocalPort );
 
-	#if( ipconfigTCP_HANG_PROTECTION == 1 )
+	/* Without bPassQueued and pxPeerSocket being set, the server socket
+	will not receive eSOCKET_ACCEPT event on child socket (instead,
+	eSOCKET_CONNECT will be used). See respective xEventGroupWaitBits()
+	call FreeRTOS_accept() and bPassQueued handling in vTCPStateChange().
+	*/
+	/* #if( ipconfigTCP_HANG_PROTECTION == 1 ) */
+	#if( 1 )
 	{
 		/* Only when there is anti-hanging protection, a socket may become an
 		orphan temporarily.  Once this socket is really connected, the owner of
-- 
2.16.2

```

forGGe, 2018-05-18:
Is it a right way to fix it?

RichardBarry, 2018-05-20:
The config file is here: https://sourceforge.net/p/freertos/code/HEAD/tree/trunk/FreeRTOS-Plus/Demo/FreeRTOS_Plus_TCP_Minimal_Windows_Simulator/FreeRTOSIPConfig.h

The patch you posted is equivalent to setting ipconfigTCP_HANG_PROTECTION to 1 in FreeRTOSIPConfig.h.  I would always recommend editing the application specific and editable configuration file rather than the source code.  In any case, when I set ipconfigTCP_HANG_PROTECTION to 0 then I can replicate the issues, so that gives me something to do on.

forGGe, 2018-05-21:
Thanks, @RichardBarry !

For now I will workaround that issue (in my project) by setting `ipconfigTCP_HANG_PROTECTION` to 1.
Looking forward for your root cause investigation.

aggarg, 2018-06-06:
I assume that setting ```ipconfigTCP_HANG_PROTECTION``` to 1 solves your problem. I am closing this issue. Let us know if you need anything else.

Thanks.

closed 2018-06-06 by aggarg