aws/amazon-freertos #32: Project cannot link when attempting to use esp_smartconfig_start()

Noggin01, 2018-05-25:
The project will compile successfully, but fails at the link stage, if I attempt to use the smartconfig feature of esp-idf.

> /home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.literal.airkiss_get_net_errno+0x4): undefined reference to `lwip_getsockopt_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.literal.KISS_Send_task+0xc): undefined reference to `lwip_htons'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.literal.KISS_Send_task+0x10): undefined reference to `lwip_socket'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.literal.KISS_Send_task+0x14): undefined reference to `lwip_setsockopt_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.literal.KISS_Send_task+0x18): undefined reference to `lwip_sendto_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.text.airkiss_get_net_errno+0x1b): undefined referenceto `lwip_getsockopt_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.text.KISS_Send_task+0x26): undefined reference to `lwip_htons'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.text.KISS_Send_task+0x7f): undefined reference to `lwip_socket'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.text.KISS_Send_task+0xba): undefined reference to `lwip_setsockopt_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_airkiss.o):(.text.KISS_Send_task+0xda): undefined reference to `lwip_sendto_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_esptouch.o):(.literal.TOUCH_Send_task+0xc): undefined reference to`ipaddr_addr'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_esptouch.o):(.text.get_net_errno+0x1b): undefined reference to `lwip_getsockopt_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_task+0x28): undefined reference to `ipaddr_addr'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_task+0x3b): undefined reference to `lwip_htons'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_task+0xcb): undefined reference to `lwip_socket'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_task+0x106): undefined reference to `lwip_setsockopt_r'
/home/hm/controller-0-10v/lib/third_party/mcu_vendor/espressif/esp-idf/components/esp32/lib/libsmartconfig.a(sc_esptouch.o):(.text.TOUCH_Send_task+0x126): undefined reference to `lwip_sendto_r'
collect2: error: ld returned 1 exit status
/home/hm/controller-0-10v/demos/espressif/esp32_devkitc_esp_wrover_kit/make/../../../../lib/third_party/mcu_vendor/espressif/esp-idf/make/project.mk:393: recipe for target '/home/hm/controller-0-10v/demos/espressif/esp32_devkitc_esp_wrover_kit/make/build/aws_demos.elf' failed
make: *** [/home/hm/controller-0-10v/demos/espressif/esp32_devkitc_esp_wrover_kit/make/build/aws_demos.elf] Error 1

I then copied the lwip component folder from the official esp-idf to the esp-idf found in a:FreeRTOS.  This yields an error stating that sys/select.h does not exit.  I then copied that file (esp-idf/components/newlib/include/sys/select.h) to the equivalent place in a:FreeRTOS.

This leads to an error, "unknown type name 'fd_set'".  fd_set is defined in esp-idf/components/newlib/include/sys/types.h, but not for the ESP_PLATFORM tool.

Is there any way to get smartconfig working on the ESP32 and using a:FreeRTOS?

sarenameas, 2018-05-30:
Hello,

ESP-IDF component *lwip* is not supported in Amazon FreeRTOS. Amazon FreeRTOS's sockets port uses the FreeRTOS+TCP stack for Espressif's ESP32-WROVER-KIT and ESP DevKitC. Since *lwip* is not supported neither is *smartconfig* that utilizes this library.

The ESP-IDF included in **lib/third_party/mcu_vendor/espressif/esp-idf** was updated to take out lwIP support for Amazon FreeRTOS. 

If you require *lwip*, then you may delete the macros surrounding the fd_set definition in types.h.
```
/* FD_SET and friends are still LWIP only */
# if !defined(ESP_PLATFORM) 
...
#endif
```

There may be other changes necessary. Please keep in mind that Amazon FreeRTOS does not support Espressif's *lwip* library, so it has not gone through the same testing as our other libraries.

Noggin01, 2018-05-30:
Is there any expectation that lwIP will be supported in the near future?  I did try deleting those macros, it led me down a rabbit hole of duplicate definitions, most of which were pretty easy to resolve.  At some point, there were a few things that were not simple to resolve.  I didn't want to spend a lot of time working on that if lwIP support is in the pipeline.

aggarg, 2018-06-06:
Amazon FreeRTOS does not support lwIP as of now. Thank you for the feedback. This is important as it helps us in prioritizing the work.

We will keep you updated if anything changes.

Thanks.

closed 2018-06-06 by aggarg