aws/amazon-freertos #33: fatal error: xtensa/tie/xt_core.h: No such file or directory

Kasuhikdas, 2018-06-06:
While Make of the project this error is thrown.

/home/kaushikdas/esp/amazon-freertos/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/include/portmacro.h:77:32: fatal error: xtensa/tie/xt_core.h: No such file or directory
compilation terminated.
/home/kaushikdas/esp/esp-idf/make/component_wrapper.mk:285: recipe for target 'lib/mqtt/aws_mqtt_agent.o' failed
make[1]: *** [lib/mqtt/aws_mqtt_agent.o] Error 1
/home/kaushikdas/esp/esp-idf/make/project.mk:467: recipe for target 'component-amazon-freertos-common-build' failed
make: *** [component-amazon-freertos-common-build] Error 2

aggarg, 2018-06-06:
Hi,

Would you please share some more details about your setup:
- Which platform are you using (Windows/Linux/Mac)?
- What is the installation path of tool chain?
- If you are using Windows, you may be hitting maximum file path length problem as mentioned here: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html. To confirm would you tell us what is the location of msys32 and Amazon FreeRTOS source code?

Thanks.

Kasuhikdas, 2018-06-06:
I am using Ubuntu 16.04.
Path is home/esp...

This is the error Everytime I try make flash.

I have just started with the Aws freeRTOS can you please guide me how can I
connect it to local mqtt broker instead of Aws.

Thanks and Regards,
Kaushik Das

aggarg, 2018-06-06:
I started with a fresh Ubuntu 16.04 machine and I was able to build by following the getting started guide here: https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html

Here is my command history:
```
sudo apt-get update  
sudo apt install build-essential  
sudo apt install gcc git wget make libncurses-dev flex bison gperf python python-serial
wget https://dl.espressif.com/dl/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
mkdir esp
cd esp
tar -xzf ~/xtensa-esp32-elf-linux64-1.22.0-80-g6c4433a-5.2.0.tar.gz
cd ..
vim ~/.profile
source ~/.profile
printenv PATH
git clone https://github.com/aws/amazon-freertos.git
cd amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/
make
```
As mentioned in the getting started guide - **"Do not install the ESP-IDF library from Espressif, Amazon FreeRTOS already contains this library. In addition, make sure the IDF_PATH environment variable has not been set."**

For connecting to Local MQTT Broker, you need to make the following changes:
- Change the broker endpoint - clientcredentialMQTT_BROKER_ENDPOINT macro in file https://github.com/aws/amazon-freertos/blob/master/demos/common/include/aws_clientcredential.h
- If your local broker uses TLS, update the keys in file https://github.com/aws/amazon-freertos/blob/master/demos/common/include/aws_clientcredential_keys.h
- If your broker does not use TLS, define the macro democonfigMQTT_AGENT_CONNECT_FLAGS to 0 in file https://github.com/aws/amazon-freertos/blob/master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/config_files/aws_demo_config.h

Hope that helps.

Thanks.

Kasuhikdas, 2018-06-07:
I worked. It was happening because of IDF path.

Kasuhikdas, 2018-06-07:
If i just put my local broker address in endpoint as you suggested will it work?

aggarg, 2018-06-07:
I am glad that it worked for you.

In order to use TLS, you need to set device certificate and key in file https://github.com/aws/amazon-freertos/blob/master/demos/common/include/aws_clientcredential_keys.h. In addition to that you need to supply the server certificate in MQTT Connect parameters (```pcCertificate``` and ```ulCertificateSize``` members of the following struct):

```c
typedef struct MQTTAgentConnectParams
{
    const char * pcURL;             /**< The URL of the MQTT broker to connect to. */
    BaseType_t xFlags;              /**< Flags to control the behavior of MQTT connect. */
    BaseType_t xURLIsIPAddress;     /**< Deprecated. Set the mqttagentURL_IS_IP_ADDRESS bit in xFlags instead. */
    uint16_t usPort;                /**< Port number at which MQTT broker is listening. This field is ignored if the mqttagentUSE_AWS_IOT_ALPN_443 flag is set. */
    const uint8_t * pucClientId;    /**< Client Identifier of the MQTT client. It should be unique per broker. */
    uint16_t usClientIdLength;      /**< The length of the client Id. */
    BaseType_t xSecuredConnection;  /**< Deprecated. Set the mqttagentREQUIRE_TLS bit in xFlags instead. */
    void * pvUserData;              /**< User data supplied back as it is in the callback. Can be NULL. */
    MQTTAgentCallback_t pxCallback; /**< Callback used to report various events. In addition to other events, this callback is invoked for the publish
                                     *   messages received on the topics for which the user has not registered any subscription callback. Can be NULL. */
    char * pcCertificate;           /**< Certificate used for secure connection. Can be NULL. If it is NULL, the one specified in the aws_credential_keys.h is used. */
    uint32_t ulCertificateSize;     /**< Size of certificate used for secure connection. */
} MQTTAgentConnectParams_t;
``` 

Also, if your broker runs on port 8883, the value of xFlags should not contain mqttagentUSE_AWS_IOT_ALPN_443:
```c
xConnectParams.xFlags = mqttagentREQUIRE_TLS;
```
You will need to change the code in function ```prvCreateClientAndConnectToBroker``` in the file https://github.com/aws/amazon-freertos/blob/master/demos/common/mqtt/aws_hello_world.c.

Hope that helps.

Thanks.

Kasuhikdas, 2018-06-07:
Did the same thing. Configured the keys also. But it is not connecting. No error while packet tracing also. Can i also put CA cert  in place of clientcredentialJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM ??

aggarg, 2018-06-07:
You should not need to change ```clientcredentialJITR_DEVICE_CERTIFICATE_AUTHORITY_PEM``` in order to connect to local broker.

Which local broker are you using? If that comes with a client, can you verify that key and certificates work with that client?

Kasuhikdas, 2018-06-08:
Thanks It worked.

closed 2018-06-08 by Kasuhikdas