aws/amazon-freertos #35: Espressif: Detected overlap at address: 0x1000 for file: .../bootloader.bin

wordyallen, 2018-06-25:
I get this error with devkitC and the wrover. I do not get this error with usinf `esp-idf`.

```
Flashing binaries to serial port /dev/cu.SLAB_USBtoUART (app at offset 0x2)...
usage: esptool write_flash [-h] [--flash_freq {keep,40m,26m,20m,80m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size FLASH_SIZE]
                           [--spi-connection SPI_CONNECTION] [--no-progress]
                           [--verify] [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x1000 for file: .../bootloader.bin
make: *** [flash] Error 2
```

huguesBouvier, 2018-06-25:
Hello,

I think you have to create a partition and flash it over the one that is causing you trouble.
The process to create and flash partitions is described here: http://esp-idf.readthedocs.io/en/latest/api-guides/partition-tables.html#.

Hugues

wordyallen, 2018-06-26:
Awesome. Thanks!

closed 2018-06-26 by wordyallen