aws/amazon-freertos #36: May I know the relationship between this repo and FreeRTOS website?

davidhuziji, 2018-07-19:
Hi,

Our previous development was based on this [FreeRTOS](https://www.freertos.org/). I'm a little confused about this repo.
May I know the purpose of this repo besides that FreeRTOS website? Is it a full package delivered as FreeRTOS kernel + diverse libraries?
Besides, may I know the difference between the FreeRTOS code in this repo and that downloaded from FreeRTOS website? Is this repo only available for official FreeRTOS code development?

If I'd like to contribute some vendor platform support, shall I continue to upload it to FreeRTOS interactive forum or shall I bring up a pull request here?

Thank you.

RichardBarry, 2018-07-19:
Hi David - The following FAQ may help to answer some of your questions - if not please ask again: https://www.freertos.org/FAQ_Amazon.html

davidhuziji, 2018-07-20:
Hi Richard,

Thanks a lot. It is quite helpful.
I have 2 remaining questions. Could you please take a look?

1. It seems that AWS FreeRTOS contains fewer board support than plain FreeRTOS package does. May I know why? Does it imply that the boards supported in plain FreeRTOS package might not be supported in AWS FreeRTOS?

2. If I'd like to contribute a board into official FreeRTOS package, can I directly upload code here for review? Or do I have to contact AWS via [https://aws.amazon.com/freertos/partners/](https://aws.amazon.com/freertos/partners/)?

Thanks a lot.

RichardBarry, 2018-07-20:
1) Working to bring more on board.  There is more to port that with just the kernel.
2) Can you please provide a little more information on which libraries are ported (just the kernel?) and to which platform - thanks.

davidhuziji, 2018-07-20:
Hi Richard,

I'd like to contribute the support to a specific development board, which is not in the demo list yet. I will not touch anything in the kernel.
I'm sorry that I cannot tell more about this board since I haven't got permission yet. I just try to understand the contribution process under AWS.

Thanks a lot.

tanmoysen, 2018-07-20:
Hi David, Could you please send an email to freertos-qual __AT__ amazon __DOT__ com with the board information and other details on your contribution? 
Thank you.

closed 2018-09-28 by dcgaws