aws/amazon-freertos #37: No OTA on the Espressif boards?

joostwestra, 2018-07-26:
Currently it is stated that the Espressif boards do not support OTA.
Is that temporary or is there a technical limitation?
What does this mean for the meaning of "Supported Hardware"? Is there a minimal subset that needs to work?

joostwestra, 2018-07-30:
Thanks for the quick reply. Does Amazon have any minimal requirements for a product to count as "Supported Hardware"?

ravibhagavandas, 2018-07-30:
There are no additional requirements if a board has already passed under Amazon FreeRTOS qualification program (AFQP). You can read more about AFQP [here](https://docs.aws.amazon.com/freertos/latest/userguide/freertos-qualification-program.html).  Using a qualified board and Amazon FreeRTOS kernel, you can write applications that can securely connect to AWS IoT.

dwlayton, 2018-08-31:
I am working on a AWS IoT application for a custom ESP32 device and will definitely need OTA support as well as BLE capability. If these features are not available in the near future for AWS freeRTOS, is the default to use AWS IoT Embedded C SDK and then use the Espressif OTA functionality?

tanmoysen, 2018-08-31:
Amazon FreeRTOS now supports OTA on ESP32. Please refer to https://amzn.to/2MEciPK for more details. Regarding BLE- while we don't yet have a standardized BLE interface for Amazon FreeRTOS, our partners such as Espressif have solutions for BLE, and our software will work with these solutions. If you want to discuss your use case further, please feel free to reach out to us at freertos-qual AT amazon DOT com.

closed 2018-10-02 by dcgaws