aws/amazon-freertos #38: ESP32 WiFi Low Power Save Modes

Sedwin97, 2018-08-01:
Board: Sparkfun ESP32 Thing
OS: Ubuntu 17.04

I'm working on research with a PhD student and I need to run an MQTT protocol with Wifi Power Save Mode on the Sparkfun ESP32 Thing board. We are using an unsecured WiFi network and simply connecting to an MQTT broker on a raspberry pi, publishing a message, and disconnecting. 

1. We were looking into the power save modes and under lib/wifi/portable/espressif/esp32_devkitc_esp_wrover_kit, we noticed that the source code returns eWifiNotSupported for the PM modes. Is this not a supported feature on the esp32 boards? And if so, why is that? The esp-idf used the power save modes.

2. We need to modify part of the U-APSD operation. I need to keep the board alive for a few ms after it would receive some data.
A similar function is shown in the image. This was used on a Cypress board using WICED and allowed us to modify the return to sleep delay of the wifi power save. I've looked all over the API but wasn't able to find anything that worked similarly. Is this a supported feature?
[image]

ravibhagavandas, 2018-08-01:
Hi,
Thanks for bringing this up. Currently the WIFI port for ESP32 board doesn't  have the power save mode support implemented. We will work on this and keep you posted once its released.
Meanwhile you can check a reference implementation of the WIFI low power APIs for TI board [here](https://github.com/aws/amazon-freertos/blob/413f30f650170e981ce470a5789b9769a42a4445/lib/wifi/portable/ti/cc3220_launchpad/aws_wifi.c#L1171). There is also a [sample ](https://github.com/espressif/esp-idf/blob/92c469b5993b001a21970f0e58b25acd3f375807/examples/wifi/power_save/main/power_save.c#L66) in esp-idf on how to set the power save mode in ESP32.

yuhui-zheng, 2019-01-31:
Since this issue has been inactive for a while now, I'm closing it. Please don't hesitate to reach out again, if any question.

closed 2019-01-31 by yuhui-zheng