aws/amazon-freertos #40: ERROR:  MQTT echo failed to connect.

alaa137, 2018-08-08:
Hi,

I'm trying to run the **FreeRTOS** example on **STM32L4 IOT** board.
I followed [this guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_st.html) and I ran the FreeRTOS sample. This is the log I'm getting:

> 0 523 [Tmr Svc] WiFi module initialized.
> 1 527 [Tmr Svc] Starting key provisioning...
> 2 531 [Tmr Svc] Write root certificate...
> 3 544 [Tmr Svc] Write device private key...
> 4 996 [Tmr Svc] Key provisioning done...
> 5 6077 [Tmr Svc] WiFi connected to AP MY_WIFI_NAME.
> 6 6082 [Tmr Svc] IP Address acquired <IP_ADDRESS>
> 7 6087 [Tmr Svc] WiFi firmware version is: C3.5.2.3.BETA9
> 8 6092 [Tmr Svc] [WARN] WiFi firmware needs to be updated.
> 9 6097 [Tmr Svc] Creating MQTT Echo Task...
> 10 6101 [MQTTEcho] MQTT echo attempting to connect to <MY_ENDPOINT>.
> 11 6283 [MQTTEcho] ERROR:  MQTT echo failed to connect.
> 12 6288 [MQTTEcho] MQTT echo test could not connect to broker.
> 13 6294 [MQTTEcho] MQTT echo demo finished.
> 14 6298 [MQTTEcho] ----Demo finished----

In the subscription topic  in AWS IoT I typed `freertos/demos/echo`as it's instructed in the tutorial...
My endpoint is correct - it's copied from `AWS Console > Settings > Endpoint` (and there it appears as `Enabled`).

What could be causing this MQTT echo error?

aggarg, 2018-08-08:
As printed in the following line, it is because of the WiFi firmware being out of date:

```
7 6087 [Tmr Svc] WiFi firmware version is: C3.5.2.3.BETA9
8 6092 [Tmr Svc] [WARN] WiFi firmware needs to be updated.
```

Please follow the instructions here in **Binary resources** section to update the WiFi firmware: https://www.st.com/en/evaluation-tools/b-l475e-iot01a.html

Thanks.

alaa137, 2018-08-12:
Hi @aggarg ,

Thanks for the response. I followed the instructions and reached step 5:
5.  Download ISM43362_M3G_L44_SPI_C3.5.2.5.STM SPI Firmware (unzip and rename .bin) from 
https://www.inventeksys.com/iwin/firmware/
I downloaded the file but what's inside the zip is already .bin so I guess no need to rename.
Then in 6 it says:
6. Copy the ISM43362_M3G_L44_SPI_C3.5.2.5.STM.bin you just downloaded to **..\Inventek FW Updater\bin folder**
But nowhere in the instructions does it say to download **Inventek FW Updater**...
Anyway I searched for it and found the following:
https://www.inventeksys.com/iwin/firmware-upgrades/
From there I downloaded **eS-Wi-Fi evaluation PC demo software** and installed it.
Still there is no **Inventek FW Updater\bin folder** like instructed in 6 above.
So I followed the video here:
https://www.youtube.com/watch?v=k4xwVNunoAw
Now I connect the board, go to _Setup > Serial Port > Configure/Open_
and select the recognized serial port (COM111, there is only one) and click OK. Here's what I get:

[image]
[image]
> The port 'COM111' does not exist.

Although the device is recognized in the device manager:
[image]

How do I proceed from here?

Thanks

alaa137, 2018-08-13:
After restarting my computer I started getting an "Access is denied" error:

[image]

aggarg, 2018-08-13:
The zip file you downloaded (named en.inventek_fw_updater, the one which contains readme.txt you are following) contains a bin directory which already has the following files:
* InventekBootloaderPassthrough.bin
* update_Wifi_FW_C3.5.2.5.STM.bat

You need to copy the downloaded ```ISM43362_M3G_L44_SPI_C3.5.2.5.STM.bin``` into this bin directory. Then you need to run the ```update_Wifi_FW_C3.5.2.5.STM.bat``` from this directory, i.e.:
```
cd bin
./update_Wifi_FW_C3.5.2.5.STM.bat
```

Let me know if that works.

Thanks.

alaa137, 2018-08-15:
That worked thank you

alaa137, 2018-08-15:
I'm actually getting the same error now although the wifi FW is updated:

```
0 521 [Tmr Svc] WiFi module initialized.
1 525 [Tmr Svc] Starting key provisioning...
2 529 [Tmr Svc] Write root certificate...
3 542 [Tmr Svc] Write device private key...
4 991 [Tmr Svc] Write device certificate...
5 1008 [Tmr Svc] Key provisioning done...
6 8169 [Tmr Svc] WiFi connected to AP SYSMOP_GUEST_2.4.
7 8174 [Tmr Svc] IP Address acquired <IP_ADDRESS>
8 8178 [Tmr Svc] WiFi firmware version is: C3.5.2.5.STM
9 8183 [Tmr Svc] WiFi firmware is up-to-date.
10 8188 [Tmr Svc] Creating MQTT Echo Task...
11 8192 [MQTTEcho] MQTT echo attempting to connect to <MY_ENDPOINT>.
12 18219 [MQTTEcho] ERROR:  MQTT echo failed to connect.
13 18224 [MQTTEcho] MQTT echo test could not connect to broker.
14 18230 [MQTTEcho] MQTT echo demo finished.
15 18234 [MQTTEcho] ----Demo finished----
```

aggarg, 2018-08-15:
Did you follow the steps here to set up AWS IoT Resources: https://docs.aws.amazon.com/freertos/latest/userguide/freertos-prereqs.html

Can you share your policy document and the region in which you are creating your resources?

alaa137, 2018-08-16:
**Policy document:**
```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": "*"
    }
  ]
}
```
What do you mean by region? In the endpoint it says **"us-east"**, and in the IoT console **"US East (Ohio)"** is picked (I didn't change that... Apparently it's default)

aggarg, 2018-08-16:
That seems correct. I'd assume that all your resources (certificates, things etc.) exist in Ohio (us-east-2). Just to be doubly sure, check that the policy is attached to the certificate you are using and the certificate is active. 

At this point, I'd have to ask you to debug this a little to help us understand the issue.
* What is the return value from the call to ```MQTT_AGENT_Connect``` in ```aws_hello_world.c```?
* Put a breakpoint in ```prvSetupConnection``` in file ```aws_mqtt_agent.c``` to see if the TCP connection is successful or not. There can be two scenarios:
    * TCP connection is successful and we are not getting the CONNACK back from the broker which would indicate that the policy is not correctly set up or timeout needs to be increased.
    * TCP connection itself fails in which case I'd put a breakpoint in the function ```ES_WIFI_StartClientConnection``` in file ```es_wifi.c``` and check the content of the buffer ```Obj->CmdData``` at line #1612 before returning from that function. 

Let me know if you want to have a remote debugging session to debug this. Apologies for the inconvenience. 

Thanks.

alaa137, 2018-08-19:
Hi @aggarg 
The return value from `MQTT_AGENT_Connect` is `eMQTTAgentFailure`.
`prvSetupConnection` completes successfully with `xStatus = 1`.
Also, in `prvSendCommandToMQTTTask` in file `aws_mqtt_agent.c`, `ulReceivedMessageIdentifier` is 34 (`eMQTTClientGotDisconnected`).

Thanks for your help

aggarg, 2018-08-20:
So it looks like TCP-TLS connection is successful but MQTT Connect is getting rejected.

Would you please enable CloudWatch logs for AWS IoT and see if we can get some useful information from there: https://docs.aws.amazon.com/iot/latest/developerguide/cloud-watch-logs.html

The AWS IoT logs in CloudWatch will have a Trace ID. Please share the trace ID and the time when you try to connect.

Thanks.

xpete, 2018-09-27:
I have the same problem but with a **ESP32** board with amazon-freertos version 1.4.1
I followed the EST32 [getting started guide](https://docs.aws.amazon.com/freertos/latest/userguide/getting_started_espressif.html).

I have the debug log enabled:
`#define configLOGGING_INCLUDE_TIME_AND_TASK_NAME    1`
`#define mqttconfigENABLE_DEBUG_LOGS 1`

This is the log i get:
>3 302 [main] WiFi Connected to AP. Creating tasks which use network...
>4 303 [main] Creating MQTT Echo Task...
>5 303 [MQTTEcho] MQTT echo attempting to connect to <MY_ENDPOINT>.
>6 303 [MQTTEcho] Sending command to MQTT task.
>7 304 [MQTT] Received message 10000 from queue.
>8 309 [IP-task] Socket sending wakeup to MQTT task.
>I (5532) wifi: pm start, type:0
>9 809 [MQTT] About to close socket.
>10 809 [MQTT] Socket closed.
>11 809 [MQTT] Stack high watermark for MQTT task: 4688
>12 809 [MQTT] Notifying task.
>13 809 [MQTT] Received message 0 from queue.
>14 809 [MQTTEcho] Command sent to MQTT task failed.
>15 809 [MQTTEcho] ERROR:  MQTT echo failed to connect.
>16 809 [MQTTEcho] MQTT echo test could not connect to broker.
>17 809 [MQTTEcho] Sending command to MQTT task.
>18 810 [MQTT] Received message 20000 from queue.
>19 810 [MQTT] Notifying task.
>20 811 [MQTTEcho] Command sent to MQTT task passed.
>21 811 [MQTTEcho] MQTT echo demo finished.
>22 811 [MQTTEcho] ----Demo finished----

This is the AWS IoT logs in CloudWatch with the trace IDs:
>{ "timestamp": "2018-09-27 14:15:54.459", "logLevel": "INFO", "traceId": "5ea7c9f5-7b42-fe24-8582-f02ed4853fcc", "accountId": "<ACCOUNT_ID>", "status": "Success", "eventType": "Connect", "protocol": "MQTT", "clientId": "iotconsole-<CLIENT_ID>-0", "principalId": "<PRINCIPAL_ID>", "sourceIp": "<IP>", "sourcePort": 64963 }

>{ "timestamp": "2018-09-27 14:15:55.331", "logLevel": "INFO", "traceId": "f16a7b27-34c7-01af-6401-baf30a1a805e", "accountId": "<ACCOUNT_ID>", "status": "Success", "eventType": "Disconnect", "protocol": "MQTT", "clientId": "iotconsole-<CLIENT_ID>-0", "principalId": "<PRINCIPAL_ID>", "sourceIp": "<IP>", "sourcePort": 64963 }

Thanks

lawrence-jeff, 2018-10-30:
I know its been a while - but you might check my issue as it is similar and you might have the same issue
https://github.com/aws/amazon-freertos/issues/74

prothorin, 2018-12-08:
[MQTTEcho] MQTT echo attempting to connect to <MY_ENDPOINT>.

MY ENDPOINT    this is strange   because in my demo write ip address or  aws iot endpoint https address.

trungnte, 2018-12-19:
I also reached the same issue. MQTT cannot hansshake to server. While script from server can do it.
The error code from TLS is -0x7280
#define MBEDTLS_ERR_SSL_CONN_EOF                          -0x7280  /**< The connection indicated an EOF. */
`6 987 [main] Key provisioning done...
7 987 [main] Creating MQTT Echo Task...
8 987 [MQTTEcho] MQTT echo attempting to connect to xxxxxxxxxxxxxxxxxxxx.iot.ap-southeast-1.amazonaws.com.
9 987 [MQTTEcho] Sending command to MQTT task.
10 988 [MQTT] Received message 10000 from queue.
11 1017 [IP-task] Socket sending wakeup to MQTT task.
I (10282) PKCS11: Initializing SPIFFS
I (10282) PKCS11: Initializing SPIFFS
I (10392) PKCS11: Initializing SPIFFS
I (10392) PKCS11: Initializing SPIFFS
I (10392) PKCS11: Initializing SPIFFS
**12 1536 [MQTT] ERROR: Handshake failed with error code -29312**

13 1536 [MQTT] About to close socket.
14 1536 [MQTT] Socket closed.
15 1536 [MQTT] Stack high watermark for MQTT task: 4052
16 1536 [MQTT] Notifying task.
17 1536 [MQTT] Received message 0 from queue.
18 1537 [MQTTEcho] Command sent to MQTT task failed.
19 1537 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
20 1537 [MQTTEcho] MQTT echo test could not connect to broker.
21 1537 [MQTTEcho] Sending command to MQTT task.
22 1538 [MQTT] Received message 20000 from queue.
23 1538 [MQTT] Notifying task.
24 1538 [MQTTEcho] Command sent to MQTT task passed.
25 1538 [MQTTEcho] MQTT echo demo finished.
26 1538 [MQTTEcho] ----Demo finished----`

mradulan, 2019-02-10:
Hi @trungnte , would you be able to run `aws int describe-endpoint` to get the information of your endpoint? Thanks.

mradulan, 2019-03-04:
closing the issue. Please reopen if the problem persists.

closed 2019-03-04 by mradulan