aws/amazon-freertos #41: Update device shadow fails

xiongyu0523, 2018-08-11:
Hi, Expert

I'm running a sample project with Amazon FreeRTOS. 
I got a problem when update device shadow when update LED status to device shadow. 
But it is always OK to update device shadow when I build a sensor packet. 

They follow the same API, the only difference is the JSON document, but I can't see problem. 

`int buildJsonAccel()
{
    /* Read data from accelerometer */
    vector_t vec = {0};
    bool read_ok = false;
    read_mag_accel(&vec, &read_ok, g_accelResolution);
    if (read_ok == false)
    {
        return -1;
    }

    /* Convert raw data from accelerometer to acceleration range multiplied by 1000 (for range -2/+2 the values will be
     * in range -2000/+2000) */
    vec.A_x = (int16_t)((int32_t)vec.A_x * g_accelDataScale * 1000 / (1 << (g_accelResolution - 1)));
    vec.A_y = (int16_t)((int32_t)vec.A_y * g_accelDataScale * 1000 / (1 << (g_accelResolution - 1)));
    vec.A_z = (int16_t)((int32_t)vec.A_z * g_accelDataScale * 1000 / (1 << (g_accelResolution - 1)));

    char tmpBufAccel[128] = {0};
    sprintf(tmpBufAccel, "{\"accel\":{\"x\":%d,\"y\":%d,\"z\":%d}}", vec.A_x, vec.A_y, vec.A_z);

    int ret = 0;
    ret = snprintf(pcUpdateBuffer, shadowBUFFER_LENGTH,
                   "{\"state\":{"
                   "\"desired\":{"
                   "\"accelUpdate\":null"
                   "},"
                   "\"reported\":%s},"
                   "\"clientToken\": \"token-%d\""
                   "}",
                   tmpBufAccel, (int)xTaskGetTickCount());

    if (ret >= shadowBUFFER_LENGTH || ret < 0)
    {
        return -1;
    }
    else
    {
        return ret;
    }
}`

`/* Reports current state to shadow */
static uint32_t prvReportShadowJSON()
{
    memset(pcUpdateBuffer, 0, shadowBUFFER_LENGTH);
    return snprintf(pcUpdateBuffer, shadowBUFFER_LENGTH,
                    "{"
                    "\"state\":{"
                    "\"reported\":{"
                    "\"LEDstate\":%d"
                    "}"
                    "},"
                    "\"clientToken\": \"token-%d\""
                    "}",
                    ledState, (int)xTaskGetTickCount());
}`
`            if (parsedLedState != ledState)
            {
                for (int i = 0; i < ledCount; i++)
                {
                    /* change from 0 to 1 */
                    if (((ledState & (1 << i)) == 0) && ((parsedLedState & (1 << i)) != 0))
                    {
                        /* turn on led */
                        configPRINTF(("Turn on %s\n", ledName[i]));
                        turnOnLed(i);
                    }

                    /* change from 1 to 0 */
                    if (((ledState & (1 << i)) != 0) && ((parsedLedState & (1 << i)) == 0))
                    {
                        /* turn off led */
                        configPRINTF(("Turn off %s\n", ledName[i]));
                        turnOffLed(i);
                    }
                }
                ledState = parsedLedState;

                /* update device shadow */
                xOperationParams.ulDataLength = prvReportShadowJSON();
                xReturn = SHADOW_Update(xClientHandle, &xOperationParams, shadowDemoTIMEOUT);
                if (xReturn == eShadowSuccess)
                {
                    configPRINTF(("Successfully performed update.\r\n"));
                }
                else
                {
                    configPRINTF(("Update failed, returned %d.\r\n", xReturn));
                }
            }
            if (parsedAccState == 1)
            {
                configPRINTF(("Update accelerometer.\r\n"));
                xOperationParams.ulDataLength = buildJsonAccel();
                xReturn = SHADOW_Update(xClientHandle, &xOperationParams, shadowDemoTIMEOUT);
                if (xReturn == eShadowSuccess)
                {
                    configPRINTF(("Successfully performed update.\r\n"));
                }
                else
                {
                    configPRINTF(("Update failed, returned %d.\r\n", xReturn));
                }
                parsedAccState = 0;
            }`

aggarg, 2018-08-14:
I tried to reproduce your problem by writing some similar code but I could not reproduce it. Would you please share a minimal working code (i.e. standalone compilable) showing the problem you are facing so that we can help you solve it. 

Thanks.

xiongyu0523, 2018-08-23:
Sorry, it was caused by hardware, turn on led will turn off WiFi. Thanks for help