aws/amazon-freertos #43: build warning on lib/FreeRTOS/tasks.c:5107

Fei-Zodiac, 2018-08-20:
When I was compiling FreeRTOS code using IAR or armgcc,I faced one warning ,which exists in task.c, "warning: 'freertos_tasks_c_additions_init' defined but not used [-Wunused-function]”

Reproduced way:
make definition for configINCLUDE_FREERTOS_TASK_C_ADDITIONS_H to 1 and no FREERTOS_TASKS_C_ADDITIONS_INIT definition 

#if( configINCLUDE_FREERTOS_TASK_C_ADDITIONS_H == 1 )

	#include "freertos_tasks_c_additions.h"

	static void freertos_tasks_c_additions_init( void )
	{
		#ifdef FREERTOS_TASKS_C_ADDITIONS_INIT
			FREERTOS_TASKS_C_ADDITIONS_INIT();
		#endif
	}

#endif

I think the “freertos_tasks_c_additions_init” definition should be wrapped by FREERTOS_TASKS_C_ADDITIONS_INIT,because this function is called only when this macro is present in the codes of other locations.

Otherwise, this makes the api freertos_tasks_c_additions_init defined but not used.

RichardBarry, 2018-08-20:
Thank you for your feedback.  Please have a look at https://sourceforge.net/p/freertos/code/2560/ to ensure you are in agreement with the change.

Fei-Zodiac, 2018-08-21:
Thank you for your reponse.It looks fine. But I care about when ,probably, will this code be merged into the main code? And in which release version will this part be updated?

RichardBarry, 2018-08-23:
It is in version 10.1.0 of the kernel here: https://sourceforge.net/projects/freertos/files/FreeRTOS/V10.1.0/

closed 2018-08-23 by RichardBarry

Fei-Zodiac, 2018-08-24:
Thanks a lot for your quick response.