aws/amazon-freertos #44: amazon-freertos does not support Will topic & message in MQTT lib

xiongyu0523, 2018-08-23:
Is it a missed feature? 
I want to use Will message to notify my application the thing is disconnected, is there a workaround?

aggarg, 2018-08-23:
Thank you for pointing it out. We will take it into consideration and prioritize it.

Thanks.

lt72, 2019-02-07:
Will message is supported with the newer lib currently in Beta in the feature/ble-beta branch. 
With the introduction of BLE support (check issue #23), we needed to re-factor the MQTT library to be transport independent and to be shareable across Amazon FreeRTOS and the [C-SDK](https://github.com/aws/aws-iot-device-sdk-embedded-C) repo as well. Please check the [ble beta branch](https://github.com/aws/amazon-freertos/tree/feature/ble-beta) in this repo and/or the [v4_beta](https://github.com/aws/aws-iot-device-sdk-embedded-C/tree/v4_beta) branch in C-SDK for the latest updates. It is a work in progress, but we'd like your feedback!

gordonwang0, 2019-05-08:
Hi everyone,

LWT support has been implemented on the release-1.5 branch.  I am closing this issue; please reopen if you have any more questions.

closed 2019-05-08 by gordonwang0