aws/amazon-freertos #45: Where to download tcptestECHO_HOST_ROOT certificate

cyliangtw, 2018-08-23:
In `aws_test_tcp.h`, default echo server address is` 34.218.25.197`, could owner provide the root certificate for secure test items such like as `AFQP_SECURE_SOCKETS_CloseWithoutReceiving` ?

aggarg, 2018-08-23:
You are expected to set up your own Echo Server as mentioned in the **Appendix J: TLS Server Setup** section of this document: https://github.com/aws/amazon-freertos/blob/master/tests/Amazon%20FreeRTOS%20Qualification%20Program%20Developer%20Guide.pdf

Thanks.

cyliangtw, 2018-08-24:
@aggarg  , I also tried to set up local TLS server with self-signed certificate based on Appendix J but Wireshark packet log showed TLS server replied “TCP Window Full”. 
The scenario as below screen-shot, device ACK window size 1000 but why the server sent “TCP window Full” ? How could I adjust local TLS server setting ?
[image]
Default TLS server `34.218.25.197` log showed the scenario is OK except unknown CA:
[image]

cyliangtw, 2018-08-24:
@aggarg ,  After enable TLS debug, log showed the failure at `"mbedtls_pk_sign() returned 6`".
Traced the source code, `rsa_prepare_blinding()` failed in private key signing.
However, in this client key exchange, the private key should be generated by random, it's "`premaster secret`". In device viewpoint, just needs to prepare correct HOST-ROOT certificate.
Is there any misunderstanding ?

dcgaws, 2018-08-24:
@cyliangtw hello, it looks like the Appendix J text that @aggarg referenced above could benefit from some improvement. In the meantime, here's a modified version of that section that attempts to be more clear about server certificate configuration versus client certificate configuration (since both are required in order for the tests to work):

**Appendix J: TLS Server Setup** 
 
A simple TLS echo server is provided with Amazon FreeRTOS code.  It is located in $AFR_HOME/tests/common/utils/tls_echo_server.go. Instructions:  

1. Install the latest version of GO on your server host: https://golang.org/dl/ 

2. Install openssl on your server host:  a. Linux --- https://www.openssl.org/source/ b. Windows --- https://slproweb.com/products/Win32OpenSSL.html 

3. Copy tls_echo_server.go to a directory you choose. 

4. Generate a TLS server self-signed certificate and private key. See $AFR_HOME/tests/common/utils/readme-gencert.txt for the openssl commands to generate a self-signed server certificate and private key.

5. Copy the server certificate and private key .pem files into a subdirectory called “certs”. The “certs” directory should be a subdirectory of the directory where the server code will run. 

6. Start the TLS server by running: go run tls_echo_server.go  

7. The server will listen on port 9000.  The IP address and the port must be set in $AFR_HOME/tests/common/tests/common/include/aws_test_tcp.h. For example if your server’s IP address is 192.168.2.6, set the following macros: 
 
Macro definition for TLS server Example value if address is 192.168.0.200 tcptestECHO_SERVER_TLS_ADDR0 192 tcptestECHO_SERVER_TLS_ADDR1 168 tcptestECHO_SERVER_TLS_ADDR2 2 tcptestECHO_SERVER_TLS_ADDR3 6 tcptestECHO_PORT_TLS         ( 9000 ) 
  
8. The tests will check the server certificate. In $AFR_HOME/tests/common/tests/common/include/aws_test_tcp.h, set tcptestECHO_HOST_ROOT_CA to your formatted server certificate. You can use the formatting tool to format your server certificate.   

9. The AFQP secure sockets tests require TLS mutual authentication to be configured. The readme-gencert.txt file also describes how to generate a client certificate and private key that is signed by the server key. This will allow the custom echo server to trust the client certificate presented by your device during TLS authentication. The client certificate and private key must be PEM formatted and copied into aws_clientcredential_keys.h before building and running the test project on the device.

cyliangtw, 2018-08-27:
@dcgaws  , thanks of your improvement of the guide, I missed the 9th step in the previous test.
I don't know which one client private key & certificate of aws_clientcredential_keys.h should be replaced by the generated client private key & certificate signed with tcptestECHO_HOST local server key.
So, I just tried to replace all of `clientcredentialCLIENT_xxx`, `tlstestCLIENT_xxx`, `tlstestCLIENT_UNTRUSTED_xxx` & `tlstestCLIENT_BYOC_xxx` in aws_clientcredential_keys.h.
However, it still failed in `"mbedtls_pk_sign() returned 6"`.

cyliangtw, 2018-08-28:
@dcgaws, after enlarge configTOTAL_HEAP_SIZE, it could pass SSL handshake without the 9th step.
Is 9th step mandatory ?
If it's, to copy the generated client certificate into which one in aws_clientcredential_keys.h ?

cyliangtw, 2018-08-29:
@aggarg , thanks of your clarification & suggestion.

closed 2018-08-29 by cyliangtw