aws/amazon-freertos #46: current shadow service code always verify clientToken, while this is optional in shadow service

xiongyu0523, 2018-08-23:
If update JSON does not contain a 'clientToken', it will get the timeout because verification of clientToken fails

aggarg, 2018-08-23:
Yes, as of now you need to put a client token in the shadow update document. As you correctly pointed out, it is optional in the AWS Shadow Service and we will see how we can make this optional in Amazon FreeRTOS.

Thank you for bringing it to our attention.

aggarg, 2018-09-07:
This is by design as of now that we do not invoke any callback if the response document does not contain a valid clientToken. We will look into how can we inform the user about the receipt of a JSON document without a valid token.

Thank you for pointing it out.

yuhui-zheng, 2019-01-31:
Appreciate bringing this up, please don't hesitate to reach out again if any other concern. Closing this issue since it has been inactive.

closed 2019-01-31 by yuhui-zheng