aws/amazon-freertos #47: quick start SetupAWS.py option to use _existing_ Thing/Policy/Cert

jrr, 2018-08-26:
This is like day two for me so I might be missing something. 

1) Set up a new device with `SetupAWS.py setup`
2) `git clean` and lose all your local state.
3) Try to `setup` again, and get errors because the Thing already exists.  Read through the other available options and scratch your head.

Is there a way to fetch details for an existing Thing, and use those?

aggarg, 2018-08-27:
As part of ```git clean```, if you have not deleted the generated certificate and private key (i.e. if you have not deleted the untracked files), you can populate the ```configure.json``` again and use the following command to configure your project:

```
SetupAWS.py update_creds
```

If you have deleted the generated certificate and private key, you need to delete the resources and recreate them using the following commands:

```
SetupAWS.py cleanup
SetupAWS.py setup
```

Thanks.

jrr, 2018-08-27:
I tried! `cleanup` seems to depend on generated artifacts:

```
jrr@jrrmbp ~/r/r/a/d/c/t/aws_config_quick_start ((413f30f6…))> python SetupAWS.py cleanup
Traceback (most recent call last):
  File "SetupAWS.py", line 193, in <module>
    cleanup()
  File "SetupAWS.py", line 147, in cleanup
    delete_prereq()
  File "SetupAWS.py", line 118, in delete_prereq
    cert_id_file = open(cert_id_filename, 'r')
IOError: [Errno 2] No such file or directory: u'john-test-thing_cert_id_file'
```

aggarg, 2018-08-27:
The current set of scripts rely on some locally generated files and we will see how we can improve them. For now, would you please delete the created certificate and policy using AWS IoT Console:

https://us-west-2.console.aws.amazon.com/iot/home?region=us-west-2#/certificatehub
https://us-west-2.console.aws.amazon.com/iot/home?region=us-west-2#/policyhub

Note that your links may vary depending on the region you are creating your resources in (change ```us-west-2``` with the region your exists in).

Then run the setup command again:

```
SetupAWS.py setup
```

Apologies for the inconvenience.

jrr, 2018-08-27:
Yeah, that's where I wound up 😒.

As you look to improve the process: it'd be great if generated sources weren't versioned / versioned sources weren't rewritten by automation.

A developer on a team, finishing up a feature branch, shouldn't have to tease out out their _code changes_ from their _I-edited-this-file-so-that-I-can-flash-my-device_ changes.

aggarg, 2018-08-27:
Thank you for your feedback. These are really good suggestions and we will take them into consideration while improving these scripts. 

Thanks.

closed 2019-01-07 by mradulan