aws/amazon-freertos #48: Arm compiler 6 support for Keil MDK 5

DavidJurajdaNXP, 2018-08-28:
I would like to kindly ask for improvement. Could you add support for ARM Compiler 6 (Clang/LLVM)?
Differences are described here: www.keil.com/appnotes/files/apnt_298.pdf

I have found following potentially conflicting cases in RVDS port:

__forceinline
        amazon-freertos\lib\FreeRTOS\portable\RVDS\ARM_CM3\portmacro.h
        amazon-freertos\lib\FreeRTOS\portable\RVDS\ARM_CM4_MPU\portmacro.h
        amazon-freertos\lib\FreeRTOS\portable\RVDS\ARM_CM4F\portmacro.h
        amazon-freertos\lib\FreeRTOS\portable\RVDS\ARM_CM7\r0p1\portmacro.h
__asm{
       amazon-freertos\lib\FreeRTOS\portable\RVDS\ARM_CM4_MPU\portmacro.h
       amazon-freertos\lib\FreeRTOS\portable\RVDS\ARM7_LPC21xx\portmacro.h

Regards,
David

RichardBarry, 2018-08-28:
Hi David - thanks for your suggestion.  I have previously had success compiling and and using the FreeRTOS GCC port with LLVM, but that was on an M0 rather than an M4F.  Did you try that?

TomasBarakNXP, 2018-09-10:
Hi Richard,

we were able to try the GCC port which seems to be compatible with the armclang v6 compiler. However it was necessary to do following modification in several files to align with the compiler version 6:
-#ifdef __CC_ARM
+#if defined(__CC_ARM) || defined(__ARMCC_VERSION)
Following files were needed to update:
demos/nxp/lpc54018iotmodule/common/config_files/FreeRTOSConfig.h
lib/include/freertos_tasks_c_additions.h
lib/third_party/pkcs11/pkcs11.h
lib/third_party/tracealyzer_recorder/streamports/Jlink_RTT/SEGGER_RTT.c
lib/third_party/tracealyzer_recorder/streamports/Jlink_RTT/include/SEGGER_RTT_Conf.h
tests/nxp/lpc54018iotmodule/common/config_files/FreeRTOSConfig.h
Other vendors' source code files are necessary to update too.

Best regards

Tomas Barak

dcgaws, 2018-09-28:
Is it possible to handle this by adding the __CC_ARM define to your build in this case, rather than taking a code change?

closed 2018-10-01 by dcgaws

DavidJurajdaNXP, 2018-11-09:
The macros discussed are predefined according to used compiler variant.
__CC_ARM - ARM Compiler 5
__ARMCC_VERSION - ARM Compiler 6 (Clang/LLVM)

Simple additional definition of __CC_ARM in  ARM Compiler 6 project could cause issues in other software components. Application determine compiler variant from this macros and it should not be manipulated.

boofish, 2019-12-04:
> Hi David - thanks for your suggestion. I have previously had success compiling and and using the FreeRTOS GCC port with LLVM, but that was on an M0 rather than an M4F. Did you try that?

Hi, Richard, how did you using the FreeRTOS GCC port with LLVM?I have trouble with that. What configuration should I make, would you give some hints?

RichardBarry, 2019-12-04:
> how did you using the FreeRTOS GCC port with LLVM

Not sure I understand the question.  I just build the source files from the GCC port layer in the LLVM project.

> I have trouble with that

What did you try?
What trouble did you have?

boofish, 2019-12-04:
> What did you try?

I try to build amazon freertos source code with llvm, generate into llvm IR code. 
In order to analysis its run time behavior through instrumentation.
 
> What trouble did you have?

I don't know how to handle it with llvm compiler to generate llvm IR code. 

I try to using command "cmake -DVENDOER=ti -DBOARD=cc3220_launchpad -DCOMPILER=arm-armclang -S . build"，but it response with error "compiler not found".

 So I need to download armclang compiler and configure it into PATH, but I don't found the compiler provider by directory 'ccstudio/ccs/tools/compiler/'，I check the directory, only found 
[image]

So, how can I do to compile the project into llvm ir code?

RichardBarry, 2019-12-04:
> So I need to download armclang compiler and configure it into PATH, but I don't found the compiler provider by directory 'ccstudio/ccs/tools/compiler/'

Did you download the compiler?  If so, which directory did you install it in?

boofish, 2019-12-05:
> Did you download the compiler? If so, which directory did you install it in?

I did not download the compiler and don't know where to download it, that's it.

boofish, 2019-12-18:
> we were able to try the GCC port which seems to be compatible with the armclang v6 compiler. However it was necessary to do following modification in several files to align with the compiler version 6.

Hi Tomas， Where did you download armclang v6 compiler, thanks.