aws/amazon-freertos #55: Complete PKCS11 Support for CC3220

tparsh, 2018-09-11:
I am having to define my own 'crypto lib' that uses the TI libraries directly as the PKCS11 crypto lib is not defined currently for the CC3220, specifically the ability to sign and verify data buffers.  Having this defined would save future developers hours of work.

gordonwang0, 2018-09-11:
Hi @tparsh,

Sorry for the inconvenience there.  We're working on getting PKCS11 fully implemented on the CC3220, however, I can't speak on a release date for it.  I'll mention this request in our planning.

yuhui-zheng, 2019-01-31:
This thread has been in active for months, that I'm closing this issue. @tparsh If you have any concern or question about PKCS11 on CC3220, please don't hesitate to reach out again.

closed 2019-01-31 by yuhui-zheng