aws/amazon-freertos #56: Improvement: Confine semantic versioning to one (or two files)

bryan-hunt, 2018-09-12:
It would be helpful for developers if the semantic version bump was confined to only a few files (e.g. VERSION, README.md).

For example in commit 8b6f1577b:
* 246 changed files with 262 additions and 252 subtractions.

However the actual code changes were:
* 3 changes in: lib/ota/portable/pc/windows/aws_ota_pal.c
* 4 changes in: tests/pc/windows/common/config_files/aws_test_ota_config.h

gordonwang0, 2018-09-12:
Hi @bryan-hunt,

Good point!  We'll take this into consideration in future releases.

RichardBarry, 2018-09-12:
Hi Bryan - I understand where you are coming from here, but personal experience shows versioning all the files in a library enables us to better support our users.

The FreeRTOS kernel in particular is on version 10.1.1 (at the time of writing), but a lot of users obtain their code from third parties, and it is not uncommon for third parties to distribute code going back to early V7.x x versions, let alone version 8 and 9.x versions.  These third parties often only ship buildable source files, and strip out the readme files, and URL links to documentation, and it is essential our users know which version they are using, and can inform us of the same when we try to help them.  I have also come across scenarios where users only update a subset of files - and support is much faster and smoother if this is immediately apparent when I look at the files.

Neither way is perfect of course, but to help I use a file diff tool that allows me to hide changes that do not effect the built image, such as comment changes.  Using that I am able to quickly identify which files have source code changes of significance (by which I mean, changes that effect the output of the compiler).

closed 2018-10-02 by dcgaws