aws/amazon-freertos #57: unreliable demo

velbn, 2018-09-12:
Hi,

I'm testing the demo for the ESP32 and I get unreliable results.
usually, I get this scenario.
-Flash the ESP32
-I get this

> I (15238) event: sta ip: 192.168.8.24, mask: 255.255.255.0, gw: 192.168.8.1
I (15238) WIFI: SYSTEM_EVENT_STA_GOT_IP
3 1478 [IP-task] vDHCPProcess: offer c0a80818ip
4 1478 [main] WiFi Connected to AP. Creating tasks which use network...
5 1479 [main] Creating MQTT Echo Task...
6 1479 [MQTTEcho] MQTT echo attempting to connect to xxxxxxxx.iot.eu-west-2.amazonaws.com.
7 1988 [MQTTEcho] ERROR:  **MQTT echo failed to connect.**
8 1988 [MQTTEcho] MQTT echo test could not connect to broker.
9 1988 [MQTTEcho] MQTT echo demo finished.
10 1988 [MQTTEcho] ----Demo finished----

-Send a reset via command line
-ESP32 reboots
-Then it works

> I (32448) event: sta ip: 192.168.8.24, mask: 255.255.255.0, gw: 192.168.8.1
I (32448) WIFI: SYSTEM_EVENT_STA_GOT_IP
3 3199 [IP-task] vDHCPProcess: offer c0a80818ip
4 3200 [main] WiFi Connected to AP. Creating tasks which use network...
5 3200 [main] Creating MQTT Echo Task...
6 3201 [MQTTEcho] MQTT echo attempting to connect to xxxxxxxxx.iot.eu-west-2.amazonaws.com.
I (39568) PKCS11: Initializing SPIFFS
I (39568) PKCS11: Partition size: total: 52961, used: 0
7 4886 [MQTTEcho] MQTT echo connected.
8 4886 [MQTTEcho] MQTT echo test echoing task created.
9 4892 [MQTTEcho] MQTT Echo demo subscribed to freertos/demos/echo
10 4896 [MQTTEcho] Echo successfully published 'Hello World 0'
11 4903 [Echoing] Message returned with ACK: 'Hello World 0 ACK'
12 5412 [MQTTEcho] Echo successfully published 'Hello World 1'

if I send again a reset command, it's a 50/50 chance if the demo will work.
It seems it needs the SPIFFS to work. But i have no idea why it initializes so slowly. Any ideas?

Thanks in advance
VELBN

JSchaenzle, 2018-09-28:
@velbn I'd be interested to see what the output looks like in the failure case if you enable the MQTT debug logs. You can do so like this: `#define mqttconfigENABLE_DEBUG_LOGS 1`

By the way, how do you "send a reset via command line"? Are you using a JTAG connection?

JSchaenzle, 2018-09-28:
Also, I can say that I've been doing quite a bit of experimenting with this demo as well and I don't recall seeing this issue.

velbn, 2018-09-28:
I don't have the time today to get the logs, but i can say. When i do "make flash monitor" and the demo fails, I press CTRL+T then R. This sends a signal to the board to reset it. I use an external USB to Serial board

velbn, 2018-10-01:
This is the logs with mqtt debug enabled
3 2683 [IP-task] vDHCPProcess: offer c0a8081aip
4 2684 [main] WiFi Connected to AP. Creating tasks which use network...
5 2685 [main] Creating MQTT Echo Task...
6 2685 [MQTTEcho] MQTT echo attempting to connect to xxxxxxx.iot.eu-west-2.amazonaws.com.
7 2685 [MQTTEcho] Sending command to MQTT task.
8 2686 [MQTT] Received message 10000 from queue.
9 2692 [IP-task] Socket sending wakeup to MQTT task.
I (27411) PKCS11: Initializing SPIFFS
I (27411) PKCS11: Partition size: total: 52961, used: 0
10 3220 [MQTT] About to close socket.
11 3220 [MQTT] Socket closed.
12 3220 [MQTT] Stack high watermark for MQTT task: 4036
13 3220 [MQTT] Notifying task.
14 3221 [MQTTEcho] Command sent to MQTT task failed.
16 3221 [MQTT] Received message 0 from queue.
15 3221 [MQTTEcho] ERROR:  MQTT echo failed to connect.
17 3221 [MQTTEcho] MQTT echo test could not connect to broker.
18 3221 [MQTTEcho] Sending command to MQTT task.
19 3221 [MQTT] Received message 20000 from queue.
20 3221 [MQTT] Notifying task.
21 3221 [MQTTEcho] Command sent to MQTT task passed.
22 3221 [MQTTEcho] MQTT echo demo finished.
23 3221 [MQTTEcho] ----Demo finished----

dcgaws, 2018-10-01:
velbn, hello, in reviewing https://github.com/aws/amazon-freertos/blob/master/demos/espressif/esp32_devkitc_esp_wrover_kit/common/application_code/main.c#L134, I believe there are some problems. For one thing, FreeRTOS_IPInit is being called before prvWifiConnect. Instead, the app start-up logic should attempt to connect to the network (and ideally confirm/wait) prior to bringing up the TCP/IP stack.

Second, the app demo task is being launched without waiting for vApplicationIPNetworkEventHook to signal that the network is up. 

A related note is that FreeRTOS+TCP depends on the PKCS#11 crypto library as a source of random numbers (RNG; including for use in DNS and DHCP packet transaction IDs, as well as selection of outbound ports). In other words, that RNG dependency will be hit early in network stack initialization. In turn, the RNG typically (including in the case of Espressif) has a hardware dependency for gathering entropy. Realistically, those factors combine to result in some delay while bringing up TCP/IP during app start-up. 

In summary, network-dependent app business logic should wait for the above to complete for launching. For the Espressif port, reordering Wi-Fi connect to happen first is the first step. Then initialize FreeRTOS+TCP and wait for its NetworkUp callback. Then run the demo task.

snap40ben, 2018-10-19:
Hi,

First boot always work fine, then if I quit miniterm and start again using **make monitor**, 50% the wifi connection will fail and display following message. 

>I (233) WIFI: SYSTEM_EVENT_STA_START
>I (353) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
>I (1333) wifi: state: init -> auth (b0)
>I (1333) wifi: state: auth -> auth (8a0)
>I (2333) wifi: state: auth -> init (2)
>I (2333) wifi: n:1 0, o:1 0, ap:255 255, sta:1 0, prof:1
>I (2333) WIFI: SYSTEM_EVENT_STA_DISCONNECTED
>1 223 [main] WiFi failed to connect to AP.

If I restart, it will work. Then will fail on next restart, etc...

Do somebody else encountered this on the aws_demo?

Thanks

iwankruger, 2018-12-19:
Hi has, has anybody been able to solve the inconsistent demo results as reported by velbn? 

Seeing similar issues.  The demo succeeds sometimes to send MQTT messages to AWS, but does fail often to connect connect to the MQQT broker.

6 1479 [MQTTEcho] MQTT echo attempting to connect to xxxxxxxx.iot.eu-west-2.amazonaws.com.
7 1988 [MQTTEcho] ERROR: MQTT echo failed to connect.
8 1988 [MQTTEcho] MQTT echo test could not connect to broker.
9 1988 [MQTTEcho] MQTT echo demo finished.
10 1988 [MQTTEcho] ----Demo finished----

sarenameas, 2018-12-22:
Please try a "make erase_flash" before "make flash monitor". This will clear any invalid certificates or other data in flash memory that may be disrupting your socket connection to the broker.

mradulan, 2019-01-07:
Hi @velbn , are you able to get reliable results on doing `make erase_flash` and then `make flash monitor`

abhidixi11, 2019-01-16:
@velbn have been unblocked now after switching to c sdk ?  Thanks

borch84, 2019-02-16:
Hi, I am implementing the demo, I am using esp32 devkitc v1. I was able to connect to mqtt broker not it is not. Here is the message from mqtt debug log:

5 848 [main] Write device private key...
6 860 [main] Creating MQTT Echo Task...
7 860 [MQTTEcho] MQTT echo attempting to connect to a3s9lktlbmjbgn-ats.iot.us-east-1.amazonaws.com.
8 860 [MQTTEcho] Sending command to MQTT task.
9 860 [MQTT] Received message 10000 from queue.
10 1861 [IP-task] Socket sending wakeup to MQTT task.
11 2361 [MQTT] About to close socket.
12 2361 [MQTT] Socket closed.
13 2361 [MQTT] Stack high watermark for MQTT task: 4696
14 2361 [MQTT] Notifying task.
15 2361 [MQTT] Received message 0 from queue.
16 2361 [MQTTEcho] Command sent to MQTT task failed.
17 2361 [MQTTEcho] ERROR:  MQTT echo failed to connect with error 1.
18 2361 [MQTTEcho] MQTT echo test could not connect to broker.
19 2361 [MQTTEcho] Sending command to MQTT task.
20 2362 [MQTT] Received message 20000 from queue.
21 2362 [MQTT] Notifying task.
22 2363 [MQTTEcho] Command sent to MQTT task passed.
23 2363 [MQTTEcho] MQTT echo demo finished.
24 2363 [MQTTEcho] ----Demo finished----