aws/amazon-freertos #58: GetIPandCertificateFromJSON couldn't find certificate

cyliangtw, 2018-09-13:
In Full_GGD test, GetIPandCertificateFromJSON got "Couldn't find certificate" error.  Details of error message as below. However, to check the capture packets, everything seems good. In GGD test, the server IP address is 18.223.9.190 & port 8443, is it the right setting ?
```
TEST(Full_GGD, GetIPandCertificateFromJSON)About to close socket.
Socket closed.
JSON parsing: Couldn't find certificate
..\..\..\common\greengrass\aws_test_greengrass_discovery.c:282::FAIL: Expected 1 Was 0
```
[image]

huguesBouvier, 2018-09-13:
Connection to greengrass is established in 3 steps:
1. It connects to you END POINT through the port 8443 (not greengrass!). From here it gets an HTML file that embbeds a JSON documents.
2. JSON document is parsed to retrieve the GG certificate, the port number then the GG IP.
3. Once IP, port number and cert are retrieved, it will connect to the GG, usually through 8883 (not 8443).

From the error message, you are able to connect but you are unable to retrieve the GG JSON file.
One potential issue is the you device police doesn't authorize GG.
For example:
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": "iot:*",
      "Resource": "*"
    },
    {
      "Effect": "Allow",
      "Action": "greengrass:*",
      "Resource": "*"
    }
  ]
}

Hugues

cyliangtw, 2018-09-14:
@huguesBouvier , my original policy is really without action for greengrass, however still got error after  I adjusted the policy with greengrass action.
Error log as:
```
TEST(Full_GGD, GetIPandCertificateFromJSON)JSONRequestGetSize: 112
SecureConnect - recv error, -30848
JSON parsing - JSON file retrieval failed

```
 Adjusted policy as:
```
{
  "Version": "2012-10-17",
  "Statement": [
    {
      "Effect": "Allow",
      "Action": [
        "greengrass:*",
        "iot:*"
      ],
      "Resource": "*"
    }
  ]
}
```

cyliangtw, 2018-09-18:
@huguesBouvier , is there any problem of my policy file for green-grass ?

huguesBouvier, 2018-09-18:
Hello,

Your policy seems to be correct. Looking at the error, it comes from an issue in the HTML document that is sent by the server.
i.e: The size is only 112. If it contains the certificate, it should be a lot bigger that that.

Could you look into the response sent by the server?
You can achieve it by adding:
'''
        GGD_SecureConnect_Read( &BufferBiggerThan112,
                                              ( uint32_t ) 112,
                                              *pxSocket,
                                              &ulReadSize );
'''

After
'''
GGD_JSONRequestGetSize( &xSocket, &ulJSONFileSize );
'''

The content of BufferBiggerThan112 will reflect what is inside the returned http page.

Hugues

cyliangtw, 2018-09-19:
@huguesBouvier ,  thanks of your helpful comment, the error message reveals the wrong region and I changed region as Oregon.
Then, to create Greengrass group, policy & core. In setting my core ->"connectivity", except to `Setting Up an Amazon EC2 Instance` by myself, is there any existing endpoint(IP or DNS) & port for test purpose ?

huguesBouvier, 2018-09-19:
Hello,

You are welcome :). 

About the test server, sadly I have never seen something like it in the GG documentation, I don't think it exists. 

Hugues

cyliangtw, 2018-09-20:
@huguesBouvier , I created green-grass group, policy, core & added device.  In my green-grass group, I created one Lambdas & local resource with status as "Affiliated". In my green-grass core, it's connectivity shows "doesn't have any endpoints configured". However, to make greengrass group a deployment, it's status seems keep `"In progress"` forever.
Based on the above condition, `GGD_SecureConnect_Read` got `{"errorMessage":"Resource not found"}` after `GGD_JSONRequestGetSize`.
Should I need to finish green-grass core's connectivity setting ?
Is there any existing shared test environment for GGD test ?

huguesBouvier, 2018-09-20:
Hello,

That's correct, you need to complete the GG procedure completetly, once deployement is complete, it should turn green "Successfully completed". 
Usual problems are:
- GG not correctly configured on the device it is running on (configure end point etc...)
- Device running GG as a firewall blocking port 8883
- In settings, add a role "Greengrass service role"

Right now there is no GGD test but I brought back the idea to our internal GG team.

Hugues

closed 2018-11-27 by dcgaws