aws/amazon-freertos #59: Add Ethernet Support for ESP32

velbn, 2018-09-14:
Hi,

I'm working on a product that needs Wifi and Ethernet connectivity.
Is it possible to add Ethernet functionality to the ESP32?

Thanks in advance!
velbn

WRansohoff, 2018-09-20:
Hello,

Unfortunately we don't currently have any plans to support Ethernet connectivity on the ESP32. Thank you for the feedback though, we always like to hear what features people are interested in seeing.

closed 2018-09-20 by WRansohoff

juanpablovega, 2018-09-29:
Ditching Amazon FreeRTOS, then.
Thanks for the fast and cold response.

WRansohoff, 2018-10-03:
My apologies for not adequately responding to your request – that was not my intent. 

We can add Ethernet connectivity to the ESP32 SoC to the FreeRTOS+TCP stack, but we do not have example code for the feature at this time. Also, if you are planning on using both Ethernet and WiFi in your application, please be aware that the FreeRTOS+TCP stack currently only supports a single interface at this time. There is an experimental version that supports multiple networking interfaces though, which you can [learn more about here](https://freertos.org/FreeRTOS-Labs/RTOS_labs_download.html).

laszlopataki, 2021-10-18:
I raised a [feature request](https://github.com/aws/amazon-freertos/issues/3409).