aws/amazon-freertos #62: Xtensa_ESP32/port.c:305 (vPortAssertIfInISR)- assert failed!

markfink, 2018-09-20:
I think I am quite close already:

* I setup AWS IOT, user, etc. and created the thing via SetupAWS.ps
* I flashed the ESP-32 and it boots

During connecting to WIFI the device rebots :(
I did not find any useful information on how to troubleshoot this issue, please advice

```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6012
load:0x40078000,len:0
load:0x40078000,len:15784
entry 0x400792e8
[0;32mI (28) boot: ESP-IDF v3.1-dev-322-gf307f41-dirty 2nd stage bootloader[0m
[0;32mI (29) boot: compile time 14:18:44[0m
[0;32mI (30) boot: Enabling RNG early entropy source...[0m
[0;32mI (35) boot: SPI Speed      : 40MHz[0m
[0;32mI (39) boot: SPI Mode       : DIO[0m
[0;32mI (43) boot: SPI Flash Size : 4MB[0m
[0;32mI (47) boot: Partition Table:[0m
[0;32mI (51) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (66) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (73) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (81) boot: End of partition table[0m
[0;32mI (85) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x11f98 ( 73624) map[0m
[0;32mI (119) esp_image: segment 1: paddr=0x00021fc0 vaddr=0x3ffb0000 size=0x023ec (  9196) load[0m
[0;32mI (123) esp_image: segment 2: paddr=0x000243b4 vaddr=0x40080000 size=0x00400 (  1024) load[0m
[0;32mI (126) esp_image: segment 3: paddr=0x000247bc vaddr=0x40080400 size=0x0b854 ( 47188) load[0m
[0;32mI (154) esp_image: segment 4: paddr=0x00030018 vaddr=0x400d0018 size=0x6d3c0 (447424) map[0m
[0;32mI (310) esp_image: segment 5: paddr=0x0009d3e0 vaddr=0x4008bc54 size=0x02574 (  9588) load[0m
[0;32mI (314) esp_image: segment 6: paddr=0x0009f95c vaddr=0x400c0000 size=0x00000 (     0) load[0m
[0;32mI (325) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (325) boot: Disabling RNG early entropy source...[0m
[0;32mI (329) cpu_start: Pro cpu up.[0m
[0;32mI (332) cpu_start: Single core mode[0m
[0;32mI (337) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (343) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (349) heap_init: At 3FFC04F0 len 0001FB10 (126 KiB): DRAM[0m
[0;32mI (356) heap_init: At 3FFE0440 len 00003BC0 (14 KiB): D/IRAM[0m
[0;32mI (362) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (369) heap_init: At 4008E1C8 len 00011E38 (71 KiB): IRAM[0m
[0;32mI (375) cpu_start: Pro cpu start user code[0m
[0;32mI (57) cpu_start: Starting scheduler on PRO CPU.[0m
I (64) wifi: wifi firmware version: f79168c
I (64) wifi: config NVS flash: enabled
I (64) wifi: config nano formating: disabled
[0;32mI (64) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
[0;32mI (74) system_api: Base MAC address is not set, read default base MAC address from BLK0 of EFUSE[0m
I (94) wifi: Init dynamic tx buffer num: 32
I (94) wifi: Init data frame dynamic rx buffer num: 32
I (94) wifi: Init management frame dynamic rx buffer num: 32
I (104) wifi: wifi driver task: 3ffc7194, prio:23, stack:4096
I (104) wifi: Init static rx buffer num: 10
I (114) wifi: Init dynamic rx buffer num: 32
I (114) wifi: wifi power manager task: 0x3ffcbdd0 prio: 21 stack: 2560
0 6 [main] WiFi module initialized. Connecting to AP HANAMI...

core=0 port_interruptNesting=1

/home/mark/esp/AmazonFreeRTOS/lib/FreeRTOS/portable/ThirdParty/GCC/Xtensa_ESP32/port.c:305 (vPortAssertIfInISR)- assert failed!
abort() was called at PC 0x40088a7f on core 0

Backtrace: 0x4008c23c:0x3ffb0770 0x4008c40b:0x3ffb0790 0x40088a7f:0x3ffb07b0 0x4008a052:0x3ffb07d0 0x400e957f:0x3ffb07f0 0x40081e92:0x3ffb0810 0x40008544:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57
...
```

WRansohoff, 2018-09-20:
Hi Mark,

To help us try to reproduce the issue, can you tell us which commands you are using to build and flash the code? And is the program that you are trying to load one of the demos in the repository?

Thanks

markfink, 2018-09-20:
@WRansohoff I just use the normal "make flash" command. The code I try to use is the default demo "mqtt" (no changes).

WRansohoff, 2018-09-20:
Okay, thank you for the information. The setup script might have already populated these values, but could you verify that your wifi credentials and MQTT endpoint are both correct in the `demos/common/include/aws_clientcredential.h` file, and that your test IoT Thing's credentials are correct in the `demos/common/include/aws_clientcredential_keys.h` file?

And for the sake of completeness, would you mind telling us what type of ESP32 development board you are using? Also, do you notice any long delays or pauses around the time when the ESP32 tries to connect to the access point or resets itself?

Thanks

velbn, 2018-09-21:
Hi @markfink 
I had a similar issue, and the problem was the power supply, it gave less voltage than indicated. I adjust correctly and the problem was gone.
A good power supply is important!
Hope it helps,
velbn

markfink, 2018-09-21:
@WRansohoff thank you for your suggestions, here the info you asked for:

The WIFI config was populated just fine.
```
/*
 * Wi-Fi network to join.
 */
#define clientcredentialWIFI_SSID       "HANAMI"

/*
 * Password needed to join Wi-Fi network.
 */
#define clientcredentialWIFI_PASSWORD   "my_secret"

/**
 * @brief Security type
 * WPA2 Security, @see WIFISecurity_t
 * Possible values are - eWiFiSecurityOpen, eWiFiSecurityWEP, eWiFiSecurityWPA,
 * eWiFiSecurityWPA2
 */
#define clientcredentialWIFI_SECURITY   eWiFiSecurityWPA2
```

About the board... it is sold by AZDelivery - unfortunately I can not say much more (I know ESP-8266 but am new to ESP-32) (https://www.amazon.de/gp/product/B071P98VTG/ref=oh_aui_search_detailpage?ie=UTF8&psc=1)

@velbn Thank you very much. I think this is a great suggestion. I try this asap and report back.

markfink, 2018-09-21:
WIFI connection works now with a dedicated power supply. Thank you so much!

WRansohoff, 2018-09-21:
Great, thank you @velbn ! I'm glad that fixed the issue.

I'll go ahead and close this, but feel free to re-open it if you have any more questions.

closed 2018-09-21 by WRansohoff