aws/amazon-freertos #63: Automatic WiFi Reconnect?

JSchaenzle, 2018-09-20:
Is there any support for handling automatic WiFi reconnect for the situation where the network drops (AP is restarted)? I've looked through the example projects and didn't see anything obvious but I just wanted to double check before I go and implement it.

WRansohoff, 2018-09-20:
Hello,

Currently, it is up to the application to manage WiFi connections and make sure that the device is still connected to a network. But if you are using the FreeRTOS-Plus-TCP library, you can define a  `vApplicationIPNetworkEventHook` callback function to detect network connect and disconnect events, as long as `ipconfigUSE_NETWORK_EVENT_HOOK` is set to 1 in your `FreeRTOSIPConfig.h` configuration file.

You can see a basic example [in the Windows simulator demo](https://github.com/aws/amazon-freertos/blob/8b6f1577b7da0712c6b8e3fdde2c18fbf1e6cf69/demos/pc/windows/common/application_code/main.c#L199). The demo does only processes `eNetworkUp` events, but you can also check for an `eNetworkDown` event in the same callback. More information can be found here:

https://www.freertos.org/FreeRTOS-Plus/FreeRTOS_Plus_TCP/TCP_Networking_Tutorial_Initialising_TCP.html

Thanks, and feel free to let us know if you have any more questions.

JSchaenzle, 2018-09-28:
Thanks for the info.

closed 2018-09-28 by JSchaenzle