aws/amazon-freertos #65: Best way to acquire date and time?

JSchaenzle, 2018-09-28:
I'm currently working on building an IoT product based on this platform and the ESP32 hardware. In my application I need to know what the current time is (actual date and time). I see that I can call `gettimeofday` from `time.h` and that returns the number of seconds since the board was powered on.  However, for that to be useful I think I need to call `settimeofday` with the number of seconds since unix epoch.  So I'm trying to work out the easiest way to get that value. It is my understanding that the device must already be getting the date and time (using sntp?) in order to handle the x.509 certificates.  If that is indeed the case, is there any way to leverage/access that datetime that is acquired in my applcation?

dcgaws, 2018-09-28:
If your device will have a reliable connection to a time server, then you could add NTP logic. However, even if you can assume that a time server is available to your devices in deployment, there are documented spoofing threats to NTP that can be tough to mitigate, especially in the context of IoT. 

The default software based implementation of TLS sockets in Amazon FreeRTOS does not perform certificate revocation checking. During TLS handshakes, it does a best-effort time validity check on server certificates based on the host image build time. Neither approach is ideal; if you can take a dependency on NTP, then I agree that it's worth considering.

JSchaenzle, 2018-09-28:
Thanks for the response. When you say, "based on the host image build time", what is the host that you're referring to?

dcgaws, 2018-10-01:
The build machine. Please take a look at https://github.com/aws/amazon-freertos/blob/master/lib/tls/aws_tls.c#L212.

JSchaenzle, 2018-10-01:
Ok. Thanks for the link. 
So does that mean that a firmware image build on this essentially has a shelf life? Do you know how long the certs that are generated by AWS are good for?

Feel free to close this issue after responding. Thanks.

dcgaws, 2018-10-01:
That code checks that the server certificate hadn't already expired by the time the firmware image was built. It's a best-effort/worst-case sort of thing.

Regarding the various root certificates used by AWS IoT servers, please see https://docs.aws.amazon.com/iot/latest/developerguide/managing-device-certs.html. 

Regarding "Thing"/client certificates issued by AWS IoT for device testing, those appear to be valid until 2049.

closed 2018-10-01 by dcgaws