aws/amazon-freertos #66: thread compile/linking/ Implementation in freertos

beikeln, 2018-09-29:
hello  Gaurav Aggarwal:
i  find the codes
in  amazon-freertos/lib/third_party/mcu_vendor/espressif/espidf/components/pthread/testtest_pthread_cxx.cpp,

beikeln, 2018-09-29:
hello Gaurav Aggarwal:
some codes  in amazonfreertos/lib/third_party/mcu_vendor/espressif/espidf/components/pthread/testtest_pthread_cxx.cpp,

gordonwang0, 2018-10-01:
Hi @beikeln,

The `thread` used in ESP's `test_pthread_cxx.cpp` is C++11's `std::thread`.  This is not supported in Amazon FreeRTOS (which is C only).

When you compile that file using `g++`, it's linking against your OS's C++ thread library.  My guess is that `arm-none-eabi-g++` does not have the C++ thread library, hence the errors.  Can you share the compilation errors you're seeing?

gordonwang0, 2018-10-29:
Hi @beikeln,

Please let me know if you need any more help with this.  Closing issue.

closed 2018-10-29 by gordonwang0