aws/amazon-freertos #68: SetupAWS.py is erroring out

JL00001, 2018-10-10:
SetupAWS.py is erroring out and not updating the header files that set things like the wifi password and ssid. 

running SetupAWS.py setup returns this error;

Traceback (most recent call last):
  File "SetupAWS.py", line 191, in <module>
    setup()
  File "SetupAWS.py", line 143, in setup
    prereq()
  File "SetupAWS.py", line 25, in prereq
    if not thing_obj.create():
  File "thing.py", line 13, in create
    assert self.exists() == False, "Thing already exists"
AssertionError: Thing already exists

and digging in to the header files, the ssid and wifi password were not updated. I am at a lost at what it means and am trying to manually edit the header files now. thanks in advance

mradulan, 2018-10-10:
Hi @JL00001 , did you check if the thing name that you created already exists in your account?

JL00001, 2018-10-10:
i do have a thing created from previous testing, would that cause a issue?

JL00001, 2018-10-10:
Ok, thanks. I didn't realize that it creates things in the AWS system and that was the "thing" that it was referring to when it said "Thing already exists" instead of the local "thing" file. I thought that the python script had something wrong with it. I thank you for your time. 

PS: maybe add a line of text in the python script to mention that.

closed 2018-10-10 by JL00001