aws/amazon-freertos #69: WiFiSeperateTasksConnectingAndDisconnectingAtOnce fail in wifi  test cases

wxiangcheng, 2018-10-12:
Hi, i did not get what this case would test. Amazon test demo spawn 2 tasks to connect 2 APs: AP1 has echo server and AP2 has not. The two tasks will check if these 2 APs have been connected. And finally check loop-back test to make sure AP1 is still connected. For our hardware device, at one time it can only connect to one AP. Is your test case ok or my understanding is wrong? If anyone can share me some information, greatly appreciated.

/**
 * @brief Spawn two threads to connect and disconnect at the same time. This is to
 * verify thread safety in the Wi-Fi driver API.
 */

alexa-noxon, 2018-10-13:
Hi,

This test is only testing thread safety (ie. that nothing catastrophic happens when 2 different threads attempt to connect to an AP at one time).

Your device only needs to be able to connect to one AP at a time, and be able to disconnect from one AP and connect to another in a threadsafe manner to pass.

Alexa

wxiangcheng, 2018-10-15:
Thanks, Alexa.
I always print below error. Is it right? This error is caused by below from WiFiSeperateTasksConnectingAndDisconnectingAtOnce -->prvConnectionTask.
/* Perform a loop-back test to ensure we are NOT connected. */
        xLoopBackResults = prvLoopBackTest();

```
_amazon: TEST(Full_WiFi, WiFiConnectMulitpleAP) PASS
amazon: Error connecting on the socket.
amazon: Error connecting on the socket.
amazon: Error connecting on the socket.
amazon: Error connecting on the socket.
amazon: TEST(Full_WiFi, WiFiSeperateTasksConnectingAndDisconnectingAtOnce) PASS_
```

wxiangcheng, 2018-10-16:
Hi Alexa,

If the device conects AP1 correctly, then thread 2 try to connect with AP2, should it be successful?
Amazon's case do in loop: 
1. WIFI_ConnectAP
2. WIFI_IsConnected
3. prvLoopBackTest
4. WIFI_Disconnect
5. prvLoopBackTest
What's the expected behaviour if 2 threads are running? Because when thread 1 does WIFI_ConnectAP succesfully, thread 2 does WIFI_ConnectAP and it will connect to another AP2. Is this right?

In case WiFiConnectMulitpleAP, it requests WIFI_ConnectAP can run twice...

sarenameas, 2018-10-17:
Hello wxiangcheng,

If the device connects to AP1 correctly, then thread 2 should have waited on a semaphore within WIFI_ConnectAP() until AP1 was finished, then thread 2 should go ahead and successfully connect. The API contract is stated in lib/include/aws_wifi.h:

 \* The Wi-Fi should stay connected when the same Access Point it is currently connected to
 \* is specified. **Otherwise, the Wi-Fi should disconnect and connect to the new Access Point
 \* specified.** If the new Access Point specifed has invalid parameters, then the Wi-Fi should be
 \* disconnected.

The device is expected to switch to the next specified access point.

wxiangcheng, 2018-10-18:
Hi Sarenameas,

Thanks. Your descripion is clear and it is what we have implemented now.
It seems there's some unexpected case in this test:
in prvConnectionTask, for each thread there's loop (5 times) and it does WIFI_ConnectAP and vTaskDelay(2) and then check WIFI_IsConnected(). When thread 1 connects AP successfully and delay 2 sec to check wifi_isConnected, thread 2 does same but it is in disconnected state-----Thread 2 would disconnect when it connects to another AP2. Then thread 1 will check WIFI as disconnected. It is not good idea to protect WIFI_IsConnected like  WIFI_ConnectAP because this is only state check. Sure, if thread 2 can connect AP2 in 2 seconds, there will be no issue here.
Did you ever have this issue? Pls correct me if my understanding is wrong.

sarenameas, 2018-10-19:
Both Thread1 and Thread2 do not call WIFI_isConnected() until both threads are synchronized and finished with calling WIFI_ConnectAP. If thread1 is finished first with calling the second WIFI_ConnectAP() it will wait for thread2 to finish calling the second WIFI_ConnectAP(). This is done with the code on line 2261 of aws_test_wifi.c:

```
        /* Wait for the other tasks to finish connecting. */
        if( xEventGroupSync(
                xTaskConnectDisconnectSyncEventGroupHandle,
                ( 0x1
                    << pxTaskParams->usTaskId ), /* Set our task ID when we are done. */
                testwifiTASK_FINISH_MASK,        /* Wait for both our task and the other
                                                  * task. */
                testwifiTASK_SYNC_TIMEOUT ) != testwifiTASK_FINISH_MASK )
        {
            configPRINTF(
                ( "Task %d timeout waiting for the other task to finish "
                  "connection.\r\n",
                  pxTaskParams->usTaskId ) );
            break;
        }
```

After the code above the WiFi should be in the connected state and both threads should return pdTRUE from WIFI_IsConnected().

wxiangcheng, 2018-10-22:
Thanks A lot. I notice that in latest version WIFI_isConnected is after xEventGroupSync. In my test version (1.2.5) it is before xEventGroupSync. I will keep it upated.

closed 2018-11-06 by wxiangcheng