aws/amazon-freertos #70: what happened to esp-idf?

markfink, 2018-10-20:
what I like about esp-idf is the build system and flexible way of adding components.

I understand that amazon-freertos is based on esp-idf but *somehow* different. Can I read up on the differences? ...and which version this is based on?

I like to add https://github.com/espressif/arduino-esp32. I put it into "AmazonFreeRTOS/lib/third_party/mcu_vendor/espressif/esp-idf/components/" since the component folder is not on top level any more.

not sure how to proceed... ideas welcome!

WRansohoff, 2018-10-22:
Hi Mark,

We use a prebuilt version of ESP-IDF in FreeRTOS, which is provided by Espressif and included as part of their port layer and GCC-based build process. So the bundled version of the framework is used both for building and flashing ESP32 binaries, and for providing a hardware abstraction layer to FreeRTOS applications.

I haven't used the Arduino-ESP32 project before, but you are correct that we have placed the ESP-IDF library files under the `lib/third_party/mcu_vendor/espressif/esp-idf/` directory. If the project that you linked to can be installed as an ESP-IDF component, then you should be able to install it normally under that directory.

And the example Makefile in the ESP32 demo project does use the built-in ESP-IDF library, so you should still be able to access specific ESP-IDF build options by running `make menuconfig` in the `demos/espressif/esp32_devkitc_esp_wrover_kit/make` directory.

Let us know if you run into any particular issues pulling that project into your application, or if you have any further questions. Good luck!

TheGreatElectron, 2018-10-25:
I am running into a similar issue. I am looking to integrate the AWS MQTT/Shadow/OTA functionality into and existing ESP-IDF project, but that has proven to be very difficult. How would one go about going from the Demo that has all of this compatibility with other boards to a clean start for production code?

WRansohoff, 2018-10-25:
The projects under the `demos/` directory are intended as demonstrations of what a new project using our libraries might look like for specific development platforms. For production code, you'll need the library files under the top-level `lib/` directory and their corresponding header files in `lib/include/`. The demos should give you an idea of how to use the libraries, and which files you will need to include. For the ESP32, a good starting point might be the source code in `demos/espressif/esp32_devkitc_esp_wrover_kit/`.

Some libraries (for example, secure_sockets) have port layers, which are different implementations of the same interface for different hardware platforms. The ESP32 port files are located under `[...]/espressif/` directories. And the AWS FreeRTOS Console does support downloading source code containing only the files which are required for a specific platform, but unfortunately the ESP32 is not yet supported in the FreeRTOS Console; we are still working on adding that support.

Keep in mind that currently, you might also need to change or remove a couple of include statements from some library files - see [issue #73](https://github.com/aws/amazon-freertos/issues/73). We are working on making it as easy as possible to add the libraries located under the top-level `lib/` directory to new or existing applications.

I hope that helps, and please do let us know if you have suggestions for improving the legibility of these libraries and demo applications.

aggarg, 2018-11-05:
Closing this for now. Let us know if you need anything else.

Thanks.

closed 2018-11-05 by aggarg