aws/amazon-freertos #73: Library dependance on aws_clientcredential.h

tombresson, 2018-10-23:
Pretty minor issue, but there seems to be dependence on `aws_clientcredential.h` in a lot of libraries, even though most of them aren't using elements defined inside of the file.

`aws_clientcredential.h` is included by the following AWS libs:
 * Defender (aws_defender.c)
 * Greengrass (aws_greengrass_discovery.h)
 * OTA Agent (aws_ota_agent.c)
 * PKCS11 (3rd party)
 * Secure Sockets (aws_secure_sockets.c)
 * TLS (aws_tls.c)

Seems like something that might already be on the radar to clean-up. It's just a bit of a hassle to keep an almost empty header around to satisfy the build of any of these libs.

WRansohoff, 2018-10-23:
Hi Tom,

Thanks for bringing this up; you're right that the configurations for individual devices are a little bit spread-out right now. Simplifying these device-provisioning settings is on our roadmap, and it is an area that we are hoping to simplify and improve. Thanks for your input, these kinds of questions help us prioritize our projects.

yuhui-zheng, 2019-01-31:
This thread has been inactive for months, cleaning up the ticket queue. Really appreciate the feedback.

closed 2019-01-31 by yuhui-zheng