aws/amazon-freertos #74: MQTT Connect Failures with 1.4.2 on ST board

lawrence-jeff, 2018-10-28:
Has something changed with the 1.4.2 recent code that would impact the online instructions? I noticed the parts about using the certificate string generating tool is no longer correct since that is not included in the AWS download anymore so was wondering if anything else has changed?

I have everything working with Mqtt-spy and can connect and publish messages to my IOT config. So I believe this validates my device/thing, certificates, policy, endpoint name and such is all correct. However the ST board refuses to connect.
I have upgraded the wifi firmware, I connect to my network just fine but can't get the board to work. Here is what I get with the debug flag set.

0 521 [Tmr Svc] WiFi module initialized.
1 525 [Tmr Svc] Starting key provisioning...
2 529 [Tmr Svc] Write root certificate...
3 542 [Tmr Svc] Write device private key...
4 985 [Tmr Svc] Write device certificate...
5 1003 [Tmr Svc] Key provisioning done...
6 3959 [Tmr Svc] WiFi connected to AP 2*******.
7 3964 [Tmr Svc] IP Address acquired 192.168.0.117
8 3968 [Tmr Svc] WiFi firmware version is: C3.5.2.5.STM
9 3973 [Tmr Svc] WiFi firmware is up-to-date.
10 3978 [Tmr Svc] Creating MQTT Echo Task...
11 3982 [MQTTEcho] MQTT echo attempting to connect to a3r******tch5-ats.iot.us-east-1.amazonaws.com.
12 3991 [MQTTEcho] Sending command to MQTT task.
13 3995 [MQTT] Received message 10000 from queue.
14 5002 [MQTT] About to close socket.
15 5011 [MQTT] Socket closed.
16 5014 [MQTT] Notifying task.
17 5016 [MQTTEcho] Command sent to MQTT task failed.
18 5021 [MQTTEcho] ERROR:  MQTT echo failed to connect...
19 5027 [MQTTEcho] MQTT echo test could not connect to broker.
20 5032 [MQTTEcho] Sending command to MQTT task.
21 5037 [MQTT] Received message 20000 from queue.
22 5041 [MQTT] Notifying task.
23 5044 [MQTTEcho] Command sent to MQTT task passed.
24 5049 [MQTTEcho] MQTT echo demo finished.
25 5053 [MQTTEcho] ----Demo finished----

I have double and triple checked the certs in the .h file and compared them to the ones working in mqtt-spy just to be sure and they are the same, the root certs in the project also appear correct.

I realize there are other similiar issues and I have tried all those fixes and recreated the setup multiple times (using various instructions). I think the Mqtt-spy setup working though validates AWS is setup ok.

I have also bumped up the MQTT timeout to 20000 ticks and it didn't help. I enabled the cloud logs and don't see any logs at the time of this debug output (although I do see connect, publish and disconnects from Mqtt-spy). Any advice?

WRansohoff, 2018-10-29:
Hi Jeff,

Sorry to hear that you're having issues. I notice that you are using an ATS endpoint; those use a different cross-chained certificate authority from the default endpoints, so those errors might be related to the root CA which is being set by default.

For now, you should be able to change the default CA used in the ST board's port file. In `[lib/secure_sockets/portable/st/stm32l475_discovery/aws_secure_sockets.c](https://github.com/aws/amazon-freertos/blob/master/lib/secure_sockets/portable/st/stm32l475_discovery/aws_secure_sockets.c#L627)`, could you try substituting the following values?

`tlsVERISIGN_ROOT_CERTIFICATE_PEM` -> `tlsATS1_ROOT_CERTIFICATE_PEM`
`tlsVERISIGN_ROOT_CERTIFICATE_LENGTH` -> `tlsATS1_ROOT_CERTIFICATE_LENGTH`

Let us know if that doesn't work, and we can continue to investigate.

WRansohoff, 2018-11-01:
Oh great, I'm glad that worked. It sounds like you probably did everything right according to the instructions, but ATS support is somewhat new in our test and demo projects. We are planning to update our code so that this extra step is not needed to use the ATS endpoints; sorry about the inconvenience.

closed 2018-11-05 by aggarg