aws/amazon-freertos #75: encoding issue with aws_iot_mqtt_publish using esp-32 @ eu-central-1

markfink, 2018-10-29:
I get "encoding" issues like the following:

``` json
{
  "format": "string",
  "payload": "\u0000\u0000{  \"temperature\": 22.97,  \"pressure\": 937.89,  \"humidity\": 30.95}",
  "qos": 0,
  "timestamp": 1540800888355,
  "topic": "farm/farm0001/sensor02"
}
```

I do not yet understand where the \u0000 comes from. Might also be an AWS issue since I started to get these once I added an iot-rule to write the messages to s3.

Here is the code to assemble and send the message:
``` c
        sprintf(cPayload,
            "{"
            "  \"temperature\": %0.2f,"
            "  \"pressure\": %0.2f,"
            "  \"humidity\": %0.2f"
            "}",
            comp_data.temperature, comp_data.pressure / 100.0, comp_data.humidity
        );
        paramsQOS0.payloadLen = strlen(cPayload);
        rc = aws_iot_mqtt_publish(&client, cTopic, strlen(cTopic), &paramsQOS0);
```

RichardBarry, 2018-10-29:
Is the message formatted correctly in cPayload?  If not, it might be the sprintf() implementation either does not, or is not configured to, support the %0.2f formatting.

markfink, 2018-10-31:
@RichardBarry thank you and yes the cPayload is formated just fine - I logged to console and checked:

``` c
ESP_LOGI(TAG, "%d - #%s#", (int) params.payloadLen, (char *)params.payload);
```

I think it was an AWS issue since I can not reproduce it anymore.

closed 2018-10-31 by markfink