aws/amazon-freertos #77: OTA failure if previous job was canceled from IoT console

nmckendree, 2018-11-01:
If active job is canceled from the IoT console then the next update will fail due to the library thinking it has an active job. 

Could this have something to do with xOTA_Agent.pcOTA_Singleton_ActiveJobName not getting cleared in the prvOTA_Close function?

OTA Library from 1.4.3.

618 144401 [OTA Task] [OTA] Flash Write 4096 @ 385024. BW: 389120 Header: 389120
619 144401 [OTA Task] [prvIngestDataBlock] Remaining: 170
620 144693 [OTA Task] [prvIngestDataBlock] Received file block 95, size 4096
621 144794 [MQTT] data sent 4, time=144794 
622 144806 [OTA Task] [OTA] Flash Write 4096 @ 389120. BW: 393216 Header: 393216
623 144806 [OTA Task] [prvIngestDataBlock] Remaining: 169

**Job was canceled**

624 144806 [OTA Task] [prvOTA_Close] Context->0x0050abf4
625 144806 [OTA Task] Sending command to MQTT task.
626 144806 [MQTT] Received message a0000 from queue.
627 144807 [MQTT] data sent 98, time=144807 
628 144862 [worker thread] Socket sending wakeup to MQTT task.
629 144862 [MQTT] Received message 0 from queue.
630 144862 [MQTT] MQTT Unsubscribe was successful.
631 144862 [MQTT] Notifying task.
632 144862 [OTA Task] Command sent to MQTT task passed.
633 144862 [OTA Task] [prvUnSubscribeFromDataStream] OK: $aws/things/A4_08_EA_D9_B9_5E/streams/AFR_OTA-dd9c502a-fc87-4a5c-b616-6da52e9de3f3/data/cbor
634 144863 [OTA Task] [OTA]	Abort - OK
635 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: execution
636 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: jobId
637 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: jobDocument
638 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: afr_ota
639 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: streamname
640 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: files
641 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: filepath
642 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: filesize
643 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: fileid
644 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: certfile
645 144863 [OTA Task] [prvParseJSONbyModel] parameter not present: sig-sha256-rsa
646 144863 [OTA Task] [prvParseJobDoc] Ignoring job without ID.
647 144863 [OTA Task] [prvOTA_Close] Context->0x0050abf4
650 200629 [worker thread] Socket sending wakeup to MQTT task.
651 200635 [MQTT] data sent 4, time=200635 

**New job is scheduled**

652 200635 [MQTT] Received message 0 from queue.
653 200636 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ jobId: AFR_OTA-110120180936 ]
654 200636 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ streamname: AFR_OTA-38b6c185-2fc0-475a-ad88-f9303ea49e65 ]
655 200636 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ filepath: / ]
656 200636 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ filesize: 1085440 ]
657 200636 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ fileid: 0 ]
658 200636 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ certfile: / ]
659 200636 [OTA Task] [prvParseJSONbyModel] Extracted parameter [ sig-sha256-rsa: e3RsCsPjxZGISL1N3UpLTDyy9DyhBC6+... ]
660 200636 [OTA Task] [prvParseJobDoc] Busy with existing job. Ignoring.
661 200636 [OTA Task] [prvParseJobDoc] Error 1 parsing job document.
662 200636 [OTA Task] [prvParseJobDoc] Rejecting job due to OTA_JobParseErr_t 1
663 200636 [OTA Task] [prvUpdateJobStatus] Msg: {"status":"FAILED","statusDetails":{"reason":"0x24000000: 0x00000001"}}

nmckendree, 2018-11-01:
Clearing xOTA_Agent.pcOTA_Singleton_ActiveJobName in the prvOTA_Close function seems to cause the selfTest mode to fail after an image is downloaded and launched. 

Maybe clear pcOTA_Singleton_ActiveJobName  somewhere else?

Clearing it definately fixes the actual issue though, I just don't know where to place it.

nmckendree, 2018-11-05:
Sorry, the Amazon FreeRTOS package is v1.4.3, i guess the OTA library is version 1.0 (from the top of the c file, Amazon FreeRTOS OTA Agent V1.0.0). It is from the latest in master from the git repository. 

Platform
My platform is a custom device board with a CYW43907 from Cypress.

aggarg, 2018-11-08:
Thank you for reporting this issue. As mentioned by @ravibhagavandas, we are able to reproduce this issue at our end and we will fix this. Meanwhile, you can apply this patch to fix the issue:
[ota_cancel_job_fix.txt](https://github.com/aws/amazon-freertos/files/2559842/ota_cancel_job_fix.txt)

Thanks.

dcgaws, 2018-12-27:
This is fixed in mainline.

closed 2018-12-27 by dcgaws