aws/amazon-freertos #87: esp32 not build error in ota

sifourquier, 2018-11-13:
Hi i go in /amazon-freertos/demos/espressif/esp32_devkitc_esp_wrover_kit/make/
and run make

the result are

/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c: In function 'aws_esp_ota_write':
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c:162:5: error: expected ';' before 'return'
     return ESP_ERR_INVALID_ARG;
     ^
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c: In function 'rewrite_ota_seq':
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c:220:42: error: 'ESP_OTA_IMG_NEW' undeclared (first use in this function)
         s_ota_select[sec_id].ota_flags = ESP_OTA_IMG_NEW;
                                          ^
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c:220:42: note: each undeclared identifier is reported only once for each function it appears in
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c: In function '_esp_get_otadata_partition':
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c:437:9: error: expected ';' before '}' token
         }
         ^
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c:438:7: error: expected '}' before 'else'
     } else {
       ^
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c: In function 'aws_esp_ota_get_boot_flags':
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c:471:14: error: 'ESP_OTA_IMG_INVALID' undeclared (first use in this function)
     *flags = ESP_OTA_IMG_INVALID;
              ^
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c: In function 'aws_esp_ota_write':
/home/simon/Sync/hepia/AWS/amazon-freertos/lib/ota/portable/espressif/esp32_devkitc_esp_wrover_kit/aws_esp_ota_ops.c:163:1: error: control reaches end of non-void function [-Werror=return-type]

Can you fix it?

alexa-noxon, 2018-11-13:
Hi, 

Are you using the IDF that comes with the Amazon FreeRTOS download?   Build errors can occur if the Espressif IDF library is installed or if the IDF_PATH variable has been set.   Also, have any of the library files been modified?  

I am not able to reproduce this build error in the OTA demo project on either Linux or Mac (I am building from 31da71acc on master and have only changed which demo is uncommented).

If you are still getting errors after confirming that the IDF_PATH is not set/the project is using the IDF in the Amazon FreeRTOS repository, please let me know about what operating system you are using and if you have made any modifications to the project.

Alexa

sifourquier, 2018-11-14:
Realy thanks you  after remove IDF_PATH that compile
but is strange to have unused source file in your git.

closed 2018-11-14 by sifourquier