aws/amazon-freertos #93: Missing include in aws_logging_task.h

Hadatko, 2018-11-15:
This header file is using data types which are not defined. Missing include file. Likely portmacro.h
https://github.com/aws/amazon-freertos/blob/master/demos/common/include/aws_logging_task.h

RichardBarry, 2018-11-15:
Thank you for taking the time to point this out.  In this case it is intended that FreeRTOS.h be included in your C file before aws_logging_task.h.  It is our preference to avoid including one header from another if possible - for example if you want to use a FreeRTOS kernel queue API then you would include FreeRTOS.h first, and queue.h (the header file that has contains the prototypes of the kernel's queue API) second.

In the case you are reporting here it is noted that a #error message in the header file informing of the need to include FreeRTOS.h first will make the code more user friendly so I will leave this issue open for now.

Just as general information - portmacro.h and FreeRTOSConfig.h should never be included in a C file directly.  Instead just include FreeRTOS.h as FreeRTOS.h ensures the other header files are brought in in the correct order.  That is one of the cases where it is necessary to have one header include another.....

Hadatko, 2018-11-15:
Well this is nonsense for me to request include some file in app in right order, instead of including it by default. For example i am using this header in c file where i don't want any another rtos related include (because why????). This is just bothering me. 
Ou it  is not working for you? it is because you forgot to include this file. We can do it for you but we will not doit.... 
You can add directly related include file there, or your common header file FreeRTOS.h...

closed 2018-11-21 by alexa-noxon